Ultimaker/Cura #13375: Cannot slice model

aIeksancler, 2022-09-24:
### Application Version

5.1.0, 5.1.1, 4.13.1

### Platform

Windows 11 10.0.22621 Build 22621

### Printer

Ender 3 v2

### Reproduction steps

1. Installed Cura from scratch.
2. Sliced model - everything worked.
3. Next day, after restarting PC - slicing does not even start and an error window pops up on 5.1.0, 5.1.1, on 4.13.1 it's just stuck at 0%.
4. Completely remove Cura with all the settings etc.
5. Go to 1.

### Actual results

Error during slicing.

### Expected results

Sliced model.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Logs after pressing "slice" button:

```
2022-09-24 21:43:49,695 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend._createSnapshot [274]: Creating thumbnail image (just before slice)...
2022-09-24 21:43:49,696 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [255]: Changing perspective mode to perspective.
2022-09-24 21:43:49,698 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.1.0\share\cura\resources\shaders\overhang.shader]...
2022-09-24 21:43:49,700 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.1.0\share\cura\resources\shaders\transparent_object.shader]...
2022-09-24 21:43:49,701 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.1.0\share\cura\resources\shaders\striped.shader]...
2022-09-24 21:43:49,751 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [289]: Starting to slice...
2022-09-24 21:43:49,751 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [308]: Going to slice build plate [0]!
2022-09-24 21:43:49,752 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2022-09-24 21:43:49,874 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [188]: Socket state changed to Listening
2022-09-24 21:43:49,890 - INFO - [MainThread] UM.Backend.Backend.startEngine [91]: Started engine process: C:\Program Files\Ultimaker Cura 5.1.0\CuraEngine.exe
2022-09-24 21:43:49,894 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [111]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura 5.1.0\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '']
2022-09-24 21:43:49,930 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [515]: Sending slice message took 0.17787981033325195 seconds
2022-09-24 21:43:49,969 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] 
2022-09-24 21:43:49,969 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] Cura_SteamEngine version main
2022-09-24 21:43:49,970 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] Copyright (C) 2022 Ultimaker
2022-09-24 21:43:49,970 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] 
2022-09-24 21:43:49,970 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] This program is free software: you can redistribute it and/or modify
2022-09-24 21:43:49,971 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] it under the terms of the GNU Affero General Public License as published by
2022-09-24 21:43:49,971 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] the Free Software Foundation, either version 3 of the License, or
2022-09-24 21:43:49,972 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] (at your option) any later version.
2022-09-24 21:43:49,972 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] 
2022-09-24 21:43:49,972 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] This program is distributed in the hope that it will be useful,
2022-09-24 21:43:49,973 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2022-09-24 21:43:49,973 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2022-09-24 21:43:49,973 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] GNU Affero General Public License for more details.
2022-09-24 21:43:49,974 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] 
2022-09-24 21:43:49,974 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] You should have received a copy of the GNU Affero General Public License
2022-09-24 21:43:49,974 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2022-09-24 21:43:49,974 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] [ERROR] Could not connect to the given address
2022-09-24 21:43:51,175 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [959]: Backend exited abnormally with return code 3221226505!
2022-09-24 21:43:51,176 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [359]: Attempting to kill the engine process
2022-09-24 21:43:51,177 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [365]: Killing engine process
2022-09-24 21:43:51,177 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [368]: Engine process is killed. Received return code 3221226505
2022-09-24 21:43:51,177 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2022-09-24 21:43:51,179 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [188]: Socket state changed to Listening
2022-09-24 21:43:51,193 - INFO - [MainThread] UM.Backend.Backend.startEngine [91]: Started engine process: C:\Program Files\Ultimaker Cura 5.1.0\CuraEngine.exe
2022-09-24 21:43:51,193 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [111]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura 5.1.0\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '']
2022-09-24 21:43:51,371 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] 
2022-09-24 21:43:51,371 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] Cura_SteamEngine version main
2022-09-24 21:43:51,371 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] Copyright (C) 2022 Ultimaker
2022-09-24 21:43:51,371 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] 
2022-09-24 21:43:51,372 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] This program is free software: you can redistribute it and/or modify
2022-09-24 21:43:51,372 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] it under the terms of the GNU Affero General Public License as published by
2022-09-24 21:43:51,372 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] the Free Software Foundation, either version 3 of the License, or
2022-09-24 21:43:51,372 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] (at your option) any later version.
2022-09-24 21:43:51,373 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] 
2022-09-24 21:43:51,373 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] This program is distributed in the hope that it will be useful,
2022-09-24 21:43:51,373 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2022-09-24 21:43:51,373 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2022-09-24 21:43:51,374 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] GNU Affero General Public License for more details.
2022-09-24 21:43:51,374 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] 
2022-09-24 21:43:51,374 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] You should have received a copy of the GNU Affero General Public License
2022-09-24 21:43:51,374 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2022-09-24 21:43:51,374 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] [ERROR] Could not connect to the given address
2022-09-24 21:43:52,601 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [959]: Backend exited abnormally with return code 3221226505!
```

aIeksancler, 2022-09-24:
The error pops up after cancelling and pressing slice button again. After initial press of the button after program startup logs shows:

`2022-09-24 22:01:41,652 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend._createSnapshot [274]: Creating thumbnail image (just before slice)...
2022-09-24 22:01:41,653 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [255]: Changing perspective mode to perspective.
2022-09-24 22:01:41,654 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.1.0\share\cura\resources\shaders\overhang.shader]...
2022-09-24 22:01:41,656 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.1.0\share\cura\resources\shaders\transparent_object.shader]...
2022-09-24 22:01:41,657 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.1.0\share\cura\resources\shaders\striped.shader]...
2022-09-24 22:01:41,735 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [289]: Starting to slice...
2022-09-24 22:01:41,736 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [308]: Going to slice build plate [0]!
2022-09-24 22:01:41,890 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [515]: Sending slice message took 0.15442585945129395 seconds
`

MariMakes, 2022-10-05:
Hey @alekseisasin,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I don't seem to have the correct information to help you.

If your slice fails on a specific model, can you share the project file?
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

If you cannot slice any models, I'll need your full logs and configuration files.
You can find your log file here:
Windows: %APPDATA%\cura\<Cura version>\cura.log or usually C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log
MacOS: $USER/Library/Application Support/cura/<Cura version>/cura.log
Ubuntu/Linux: $USER/.local/share/cura/<Cura version>/cura.logs

You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Thank you 👼

aIeksancler, 2022-10-11:
> C:\Users\\AppData\Roaming\cura<Cura version>\cura.log
> MacOS: $USER/Library/Application Support/cura//cura.log
> Ubuntu/Linux: $USER/.local/share/cura//cura.logs

Hello!

I cannot slice ANY model. I've cleared the logs and saved only what happens
after I press slice. Nonetheless, full logs are in the zip in attachement.

PS I've installed 5.2.0-beta a second ago. So far it is slicing. =)

BR,
Aleksander

MariMakes, 2022-10-11:
Hey @aIeksancler,

Do I understand correctly that your issue is resolved?
Can I close the issue?

closed 2022-10-25 by github-actions[bot]