Ultimaker/Cura #13376: Failed to slice

CogworkGoblin, 2022-09-25:
### Application Version

5.1.1

### Platform

Windows 10

### Printer

anycubic Kobra

### Reproduction steps

attempting to slice for first time. got to about 10% and failed no more info rprovided

### Actual results

failed - unexpected error

### Expected results

slice as per normal

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

no

GregValiant, 2022-09-25:
Without a project file or the model we can't really tell what's going on.
Have you tried moving and/or rotating the model on the Cura build plate?  Sometimes that can fool the bug that might be keeping it from slicing.  It may be a model error though.  Without more info there is no way to tell.

CogworkGoblin, 2022-09-25:
thanks went back to older version with success.

see attached file
 sky-islands-6x6-isle-lighthouse-base.stl
<https://drive.google.com/file/d/1tuv-jInAIJRwgBp1R9EtYCLhB-SDyg2F/view?usp=drive_web>

GregValiant, 2022-09-25:
I don't have permission to download from there.  Can you zip the file and post it here?

CogworkGoblin, 2022-09-25:
[sky-islands-6x6-isle-lighthouse-base.zip](https://github.com/Ultimaker/Cura/files/9640926/sky-islands-6x6-isle-lighthouse-base.zip)

GregValiant, 2022-09-26:
Thanks for the file.
There are 4 small errors (one at each corner of the base) according to Mesh Mixer and 3D build didn't find any errors.  This model should slice for you.  I didn't have any trouble slicing it on Windows 10 and Ender 3 Pro.
This type of model (high-resolution with no errors) has been trouble in Cura for quite a few people and the bug seems to be Windows specific.  As I said above - moving and/or rotating the model seems to fool the bug.  That doesn't always work bt it's worth a try.
I believe this bug has been worked on by the Cura team and should be fixed in 5.2.
I'll mark this as a duplicate and a slicing

CogworkGoblin, 2022-09-26:
Awesome, thanks!
a slight rotation seemed to do the trick!