Ultimaker/Cura #13378: Won't open 3mf files when using dual monitors

andyrblank, 2022-09-25:
### Application Version

5.1.1

### Platform

Windows 11

### Printer

Wanhao Duplicator i3 (MP Maker Select V2)

### Reproduction steps

1. Open Cura
2. Open 3mf file

### Actual results

Cura locks up and becomes unresponsive and UI is blank.
[image]

### Expected results

Cura should open the 3mf file with dialog
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I am able to get Cura 5.1.1 to open a 3mf file if I disable one of my monitors in my dual monitor setup. Cura 4.13.1 works to open the same 3mf file regardless of monitor configuration. I have also tried saving a 3mf file from Cura 5.1.1 and opening the same file and it had the same issue when two monitors are running.

System Info:
Windows 11 Build 21H2
Alienware/Dell OEM Nvidia Geforce 1660 TI
Nvidia Game Ready Driver 516.94
Intel Core i5 9400
32GB RAM
[WDI3_Dragon_Skull_Back_3DKitbash.zip](https://github.com/Ultimaker/Cura/files/9641522/WDI3_Dragon_Skull_Back_3DKitbash.zip)

andyrblank, 2022-09-25:
The same issue occurs with a clean install of the Nvidia Studio driver 517.40.

andyrblank, 2022-09-25:
I had been searching this in the issues as a 3mf issue but forgot to search again once I knew it was related to dual monitors. Now that I have I found information in this thread https://github.com/Ultimaker/Cura/issues/12197 that suggested stopping the `Nahimic Service`. Once I did that the 3mf files open without issue with dual monitors.

andyrblank, 2022-09-26:
Cura Log:
[cura.log.zip](https://github.com/Ultimaker/Cura/files/9642665/cura.log.zip)

fieldOfView, 2022-09-26:
As you have found, this is a duplicate of #12197. The issue is not specific to this particular dialog, but affects many parts of Cura that open dialogs.

closed 2022-09-26 by fieldOfView