Ultimaker/Cura #13386: Does not save G code correctly to device

TheLaw23, 2022-09-26:
### Application Version

5.1.1

### Platform

chrome os

### Printer

ender 3

### Reproduction steps

I had downloaded Cura 5.1.1 with no issues to my Lenovo chromebook C340-11 and have been using no problem until i go to download my G code. It downloads as seen from videos pulling up in linux but instead of G code it is a readable file that i can open.

### Actual results

computer unable to read as G code.

### Expected results

It will not allow me to drop file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I TRIED.

GregValiant, 2022-09-26:
Thanks for the report.
Github will only accept ZIP files and *.txt files so you need to put your files into a zip folder and post the zip here.

A gcode file is indeed a simple text file and will open in Notepad or any other text editor.  The beginning of a Cura gcode should look something like this:

;FLAVOR:Marlin
;TIME:24051
;Filament used: 121.781m
;Layer height: 0.5
;MINX:3.001
;MINY:28.002
;MINZ:0.5
;MAXX:226.998
;MAXY:201.998
;MAXZ:75
;Generated with Cura_SteamEngine main
M82 ;absolute extrusion mode
;
; Greg's E3Pro StartUp
;
M82
G90
G21
M140 S60
M109 S200
G92 E0 ; Reset E
G28 X Y; Home XY axes

That's the language your Ender understands.
Zip those files (right click, select "Send To" and then "Compressed (Zip) Folder") and post them here.
I'll leave this as a bug report for now but it sounds like it is doing as expected.

TheLaw23, 2022-09-28:
Yes I understand that you can edit Gcode and it's instruction configured
language. But it is exporting those instructions as a text file type in the
file description instead of a Gcode file. When I open my files it is listed
what is Gcode and what is not. I cannot open a Gcode file directly from
files, this file it is exporting I can. It is not readable by my printer.

GregValiant, 2022-09-28:
Please follow the instructions in my previous post, zip the file, and post it here.  I need to see what you have there in order to figure out what is going on.

TheLaw23, 2022-09-29:
[Example.zip](https://github.com/Ultimaker/Cura/files/9678082/Example.zip)

TheLaw23, 2022-09-29:
There it is just downloaded and zipped nothing else

GregValiant, 2022-09-30:
That's a proper gcode file but with no filename extension.  Maybe @MariMakes has a solution.  So far as I know Cura always adds the gcode extension.

TheLaw23, 2022-09-30:
hopefully because i cant print out anything using cura until its fixed

GregValiant, 2022-09-30:
In Windows Explorer right click on the file.  Select "Rename" and add the .gcode extension.  Dismiss the Windows info box that comes up.  Try to print the file.
The inside guts of the file is fine, as far as I can tell it's just missing the .gcode extension.

TheLaw23, 2022-09-30:
i'm using linux through chrome os same process?

TheLaw23, 2022-09-30:
I don't believe i can all im able to do is rename its search label nothing else

TheLaw23, 2022-09-30:
Never mind!!! Thank you for the help it fixed with a ".gcode" added to the
name!

MariMakes, 2022-09-30:
Hey hey. 

I'm sorry I couldn't help, but I'm super happy that you managed to find the answer. 🎉 
Thanks for helping out @GregValiant 

We are still investigating ways to support chrome OSses better. 
Reports like this really help. Thank you 😄 

I'll close this issue since it's resolved.

closed 2022-09-30 by MariMakes

Marissamarie, 2022-10-29:
> Never mind!!! Thank you for the help it fixed with a ".gcode" added to the name!
> […](#)

Hey! Can you put an example of how you did that? I am having the same issue. it's opening a different-looking window than my original files. I appreciate your help! :)