Ultimaker/Cura #13387: Shape position not saved/loaded correctly when using 3MF file

rondlh, 2022-09-27:
### Application Version

5.1.1

### Platform

Windows 10

### Printer

Anycubic i3 Mega

### Reproduction steps

1. Import a model
2. Add some parts (cube/cylinder) from the "Parts for calibration" plugin
3. Place the added parts at specific locations
4. Save the project as a 3MF file
5. Close Cura
6. Load the 3MF file (Open as project)

### Actual results

All shapes (cubes/cylinders) except the main shape have moved (Y position is reduced by 10 mm).

### Expected results

Reload the shapes in the original position.

[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

A picture of a cube before and after reloading the project.
The car position stayed the same, but the cube and cylinders Y position was reduced by 10 mm

MariMakes, 2022-10-11:
Hey @rondlh,

Thanks for your report 👍 
Sorry, it took us a while to get back to you 😞 

Can you check for me in your preferences if you have these settings checked? 
They influence the position of the meshes that you load from a project file.

If this is not the cause, can you share your project file with us? 
It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

rondlh, 2022-10-15:
> Hey @rondlh,
>
> Thanks for your report 👍 Sorry, it took us a while to get back to you 😞
>
> Can you check for me in your preferences if you have these settings checked? They influence the position of the [...]

Thanks, I have both options switched off. The problem is in the Y-direction, so dropping to the build plate should not cause this issue. I'm happy to share the project. Currently when I load the project, I add 10mm to the Y-position of the Cube and 2 cylinders to correct the issues I have. (Cube: Y=14.745mm, Cylinder: Y=22mm, Cylinder(1): Y=22mm). These parts are from the "Part for calibration" extension.
[Suzuki_Jimny-Print_in_Place3.zip](https://github.com/Ultimaker/Cura/files/9791355/Suzuki_Jimny-Print_in_Place3.zip)

MariMakes, 2022-10-17:
Hey @rondlh,

Thanks for sharing your project file. 
I can reproduce it on my side and it's indeed super odd 😮 

I've discussed it to the team and we've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9755

Thanks for the report! 👍 

I also think it might be connected to https://github.com/Ultimaker/Cura/issues/13523

rondlh, 2022-10-19:
> Hey @rondlh,
>
> Thanks for sharing your project file. I can reproduce it on my side and it's indeed super odd 😮
>
> I've discussed it to the team and we've added a ticket to the backlog with the intent [...]

Thanks, it's good to hear that the issue can be reproduced. I just tried it on Cura 5.2.1, and the issue still remains.

rogerw-gh, 2023-01-11:
Same problem observed on Ubuntu LTS 20.04.05, however unchecking Keep Models Apart leaves the models where they were positioned when the 3MF was created by Cura 4.12.1

saumyaj3, 2023-10-13:
hey @rondlh 

I was analysing the issue: 
It seems like the parts (cube/cylinder) from the "Parts for calibration" plugin are to be used in a group. Is it possible if you can guide me to that plugin?

rondlh, 2023-10-13:
OK, that's good to know, thanks a lot for the efforts!

closed 2023-10-13 by rondlh