Ultimaker/Cura #13393: On Linux, Cura 5.1.0 crashes when a file dialog should be shown

JohnCalin, 2022-09-28:
### Application Version

5.1.0

### Platform

Gentoo Linux, nvidia proprietary driver

### Printer

Anycubic Mega X

### Reproduction steps

1. launch Cura 5.1.0
2. perform any function which should present a file dialog (for example, file->open files)

### Actual results

Cura crashes immediately.
Console displays:
```
(Ultimaker-Cura:12924): Gtk-WARNING **: 20:45:41.881: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Adwaita/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Adwaita/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
/tmp/.mount_UltimaCaNalH/AppRun: line 16: 12924 Aborted                 $scriptdir/Ultimaker-Cura "$@"
```

### Expected results

File dialog should have appeared

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9661324/cura.log)

JohnCalin, 2022-09-28:
For what it's worth, I _am_ able to use Cura so long as I avoid file dialogs.  By drag-and-drop I am able to load STL files, and they slice and print just fine.

I am not able to save a project, since there's no way to do that without a file dialog.

ckvsoft, 2022-09-28:
It doesn't seem to affect all distributions though. It works for me on kubuntu without any problems

JohnCalin, 2022-09-29:
The path to `image-missing.png` in the error message changes if I change my icon theme (though the error happens with every theme I have installed).  @ckvsoft  which icon theme are you using?

JohnCalin, 2022-09-29:
Also, since it may matter, I'm using the XFCE desktop.

ckvsoft, 2022-09-29:
I use breeze and als
application style fusion.
I had a similar problem with FreeCad's Appimages. It didn't crash, but I didn't have any icons in the file dialog selection. This works with application style Fusion.
Are you using the "modern" appimage?

JohnCalin, 2022-09-30:
I'm not sure how to answer that question.
The filename is Ultimaker-Cura-5.1.0-linux.AppImage

JohnCalin, 2022-09-30:
I just saw 5.2.0-beta is available, so I downloaded.  I see now what you meant about the 'modern' appimage.
The crash still happens with the non-modern appimage.  Using modern, there is no crash but the file dialogs do not display.  No errors are shown on the console.

MariMakes, 2022-10-11:
Hey All,

Sorry, it took us a while to get back to you 😞 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/12266
We have a ticket on our backlog with the intent to improve this behavior.
If you read through the thread you'll see a couple of suggestions as work around. 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12266

closed 2022-10-11 by MariMakes

FlorianBador, 2023-04-23:
6 months later using the 5.3.0 version I am experiencing the exact same issue (a total crash, not just the dialog that doesn't open).
I also run XFCE window manager (on Slackware).
I didn't find any solution so I am not able to use Cura anymore.

MariMakes, 2023-05-04:
Hey @FlorianBador,

We might have a fix for this issue. 🎉
It's a bit hard for us to test, but we have a nightly build available.

Can you please try it out and let us know if it resolves your issue?
You can find the download links and the discussion here: https://github.com/Ultimaker/Cura/discussions/15098

strangeattraction, 2023-06-06:
Hello
I'm using cura 5.3.1 on Manjaro and still have similar issue - i trying to save my project, but file dialog doesn't appears

MariMakes, 2023-06-06:
Hey @strangeattraction,

Did you get a chance to check the special version of Cura that I mentioned above? 
You can find the download links and the discussion here: https://github.com/Ultimaker/Cura/discussions/15098

If it doesn't resolve your issue, feel free to create a new issue to discuss what is happening.

strangeattraction, 2023-06-06:
hello @MariMakes 
i downloaded [UltiMaker-Cura-5.4.0-alpha-cura_10389_148-linux-modern.AppImage](https://storage.googleapis.com/software.ultimaker.com/cura_nightlies/UltiMaker-Cura-5.4.0-alpha%2Bcura_10389_148-linux-modern.AppImage) and i got an error:

> libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_UltiMahyXwvo/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/dri, suffix _dri)
> libGL error: failed to load driver: iris
> libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_UltiMahyXwvo/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/dri, suffix _dri)
> libGL error: failed to load driver: iris
> libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_UltiMahyXwvo/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/swrast_dri.so) (search paths /usr/lib/dri, suffix _dri)
> libGL error: failed to load driver: swrast

finnfabians, 2023-08-10:
this is still happening (Arch linux)

MariMakes, 2023-08-15:
So I'm a Windows user but I've been told that you can try to `apt install xdg-desktop-portal-gtk`

You can set the `QT_QPA_PLATFORMTHEME `to one of the following values: 
- gtk2
- gnome
- kde
- qt5ct
- xcb
- xdgdesktopportal (current default).

Or start the Cura AppImage with the argument `-platformtheme ` with one of the following:
- gtk2
- gnome
- kde
- qt5ct
- xcb
- xdgdesktopportal (current default).

There could be more, possible values are scattered over the internet; Feel free to complete this list.

Can you let us know if that works for you? 

If that's not the case please let us know and we'll be happy to take another look.