Ultimaker/Cura #13395: Version 5.1 produces too thick inner walls

vikmaks1, 2022-09-28:
### Application Version

5.1.0

### Platform

Win10

### Printer

-

### Reproduction steps

Starting from the 5.1.0 version there is a strange behavior with slicing models which have inner (crossing) walls like in the attached example  - any wall that is between outer walls will be sliced thicker than it needs to be. The same slicing settings were used in both 5.0 and 5.1 version.

### Actual results

Model has only 1mm walls, but the middle crossing wall is sliced thicker where it is between outer walls, also there are some width artifacts in the middle section:
[image]

### Expected results

The same stl model with Cura version 5.0 slices correctly: 
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[VK_test.zip](https://github.com/Ultimaker/Cura/files/9668573/VK_test.zip)

MariMakes, 2022-10-11:
HEy @vikmaks1 

Welcome to the Ultimaker Cura Github 🚀 

Thanks for your report, this is a super interesting observation.
In Cura 5.1 we improved the toolpath simplification algorithm. The algorithm that reduces the resolution of the print has been rewritten, providing a more constant resolution to allow for greater precision without causing buffer underruns.

I believe what you are seeing is expected behavior, but I'll bring it up to the team just to be sure. 

I've noticed that if you alter your extrusion Area Deviation to a very low value, you'll have the behavior back that you are looking for. 
[image]

vikmaks1, 2022-10-16:
Thank you for looking into this weird issue! Indeed, bringing Area Deviation down to a very small value makes lines matching the model, sadly this workaround is inconsistent across different models - initially provided model is 'fixed' by lowering to 50 um^2, but more complex models like the one attached here ends up with inconsistent width artifacts: 
[image]
Lowering the value decreases these inconsistencies until they are gone at around 10 um^2, but will it have the same effect on other models remains a question.

More complex model project file: [VK_14.3x8.3-3X10.zip](https://github.com/Ultimaker/Cura/files/9794091/VK_14.3x8.3-3X10.zip)

MariMakes, 2022-10-26:
I've brought your issues up to the team and there is something odd indeed. 
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9798

Thanks for the report! 👍