Ultimaker/Cura #13397: Failed to slice

Robandhooch, 2022-09-28:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Anycubic Vyper

### Reproduction steps

Loaded file as usual
Sliced
Get error message. Failed to slice

Same file on my Windows 10 desktop with same profile slices just fine.

Been using this combo of OS, Slicer version, and file creator for months without this issue.
Been using cura for several years, first time i've seen this issue.

### Actual results

Nothing, just error message

### Expected results

Expected results, file sliced and goes into printer, and prints fine,
Actual results, nothing but error message

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[AV_LVL2_Eastwing_BedroomA FAILED TO SLICE.zip](https://github.com/Ultimaker/Cura/files/9669518/AV_LVL2_Eastwing_BedroomA.FAILED.TO.SLICE.zip)

Robandhooch, 2022-09-28:
no idea how to upload a log file

Robandhooch, 2022-09-29:
Restarted computer, no change
Installed 5.1.1, no change

MariMakes, 2022-10-13:
Hey @Robandhooch 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13500
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13500

closed 2022-10-13 by MariMakes