Ultimaker/Cura #13400: Disallowed areas vs build plate adhesion

0r31, 2022-09-29:
### Application Version

5.1.1

### Platform

Windows 11 Pro

### Printer

Any

### Reproduction steps

Hi all,

When a printer has disallowed areas defined, changing build plate adhesion from anything to None clears the disallowed areas on the scene view. Adding an object to print make disallowed areas appear again.

Default build plate adhesion is brim:
[image]

Setting build plate adhesion to None:
[image]

Adding an object to print:
[image]

### Actual results

The disallowed areas disappear on the scene view after setting build plate adhesion to None.

### Expected results

The disallowed areas shouldn't disappear on the scene view after setting build plate adhesion to None.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

It is a very minor issue, no need to fix it asap.

GregValiant, 2022-09-29:
Thanks for the report.
Here is how it looks in my setup.  This is with no model, 15mm brim, and 50mm X offset between the extruders.
[image]

This is with the model set to extruder2 and the brim on extruder1.
[image]

So I am unable to duplicate the problem.  Please post a project file (File | Save Project.  When I open the 3mf file it will have your printer and settings.  I can throw out the model and start fresh and hopefully see what is going on.

0r31, 2022-09-29:
Hi @GregValiant ,

Thanks for your tests!

Could you please select a "None" build adhesion plate after your first screenshot (and so without any object to print)?

Regards,

GregValiant, 2022-09-30:
[image]

That image is with a dual extruder Ender 3 Pro and the Ender 3 Pro definition does not have defined disallowed areas.  

I had to go back to 4.8 as I hadn't altered the Ender 3 definition file in that version.  A dual extruder Ender 3 with a defined disallowed area in the front and rear (for the glass clips) looks like this without a model:
[image]

And like this with a model:
[image]

So I guess I can see your point but I don't see it with "any" printer but as you say, with printers that have defined Disallowed Areas in their definition files.
I think there may be other issues with Cura that will take priority over this.  I don't see this as a bug myself because without a model to slice it would seem that whether or not disallowed areas are displayed doesn't matter but I'll leave the bug label on this for you.  Someone from the Cura team will take a look.

0r31, 2022-09-30:
As i said, very minor bug, no impact on gcode generation, so could be fixed ... one day (or never).