Ultimaker/Cura #13403: Try to slice a file  and send me to this url

adlsistemas, 2022-09-29:
### Application Version

5.1

### Platform

windows 11

### Printer

Ender 3

### Reproduction steps

1 Import file
2 Click button slice

### Actual results

Show error "slicing  failes"

### Expected results

slice the model

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[STL file.zip](https://github.com/Ultimaker/Cura/files/9678764/STL.file.zip)

GregValiant, 2022-10-02:
Thanks for the report.
Here is a report from OnlineSTLRepair on your model.
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 12 Non-manifold edges (?)
--> 17 Inverted faces (?)
--> 2 Degenerate faces (?)
--> 10 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 99955ms ------
-> Vertex count changed from 57750 to 57680 (-70)
-> Triangle count changed from 116298 to 116156 (-142)
Cura 5.0 and 5.1 have a bug that prevents certain models from slicing.  Moving and /or rotating the model on the build plate will often fool the bug.  I was able to slice your model with both 5.1 and 5.2beta.
Areas of that model don't look good in the preview even when I sliced with a 0.2 nozzle.  In addition, because the "hands" extend 1mm below the rings, the entire model needs support.  If the latticework was to print, there is the problem of removing support from such a delicate structure.
Here are two of the pivot pins.  You can see that the upper one is only attached by a very small contact patch.
[image]

Here is the lower "hand" and it has issues.  This is with a 0.2 line width and .1 layer height.  The index and middle fingers aren't really attached.
[image]

I'll leave this as a bug report for you.  As it is a duplicate report regarding the slicing in 5.1, I'll mark this as a duplicate and close it in favor of the previous reports.  I hope you understand.  Going beyond the bug though...that model has issues.

closed 2022-10-02 by GregValiant

adlsistemas, 2022-10-05:
Hi, thanks for the feedback, I'm still learning, could you tell me how you analyzed it?

GregValiant, 2022-10-06:
Number one is that any lattice is tough to print and the finer the lattice, the tougher it is.  There are lots of small areas that often are below the Line Width and so they get left out.  Some models are obviously (to me) created from a cartoon or magazine character and were generated in some kind of software.  Those often have errors as the artist may not be familiar with what makes a printable model (as opposed to an impossible model).  Then there are models that were created by scanning an existing physical object.  Scanning creates a "cloud" of millions of points reflected from the surface of the object.  Then another piece of software takes those millions of points and makes a model.  Amongst the millions of points what are the chances that they were all perfect?  The sheer density of data in a point cloud can bring a computer to it's knees begging for mercy.  It's a lot for a computer to deal with much less and Iphone or Google tablet.
When you get models off the internet always check them for errors before you spend time trying to slice.  Programs like MS 3D Builder (which is included with Windows) and Mesh Mixer are pretty good.  Cura's own Mesh Tools plugin (available in the MarketPlace) is good for checking a model but so-so at actual repairs but Mesh Tools is right there with a right click so it's very handy.  That on-line site I mentioned is pretty good so long as the model file is less than 20mb.

adlsistemas, 2022-10-06:
Thanks for the tips and help.

I will continue learning more about modeling