Ultimaker/Cura #13405: Slicing error

GenesioRodrigues, 2022-09-29:
### Application Version

5,1,0+Beta304

### Platform

Windows 10

### Printer

Two Trees Saphire Plus

### Reproduction steps

1. after click "Slice"button

### Actual results

App has oppend a popup to report a bug in clicing

### Expected results

object sliced

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9679293/cura.log)
[TSAB_Bastao30cm4PartesYredBUG.zip](https://github.com/Ultimaker/Cura/files/9679295/TSAB_Bastao30cm4PartesYredBUG.zip)

MariMakes, 2022-10-25:
Hey @GenesioRodrigues 

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I can recreate your crash on my side, so I'll bring it up to the Cura team.
I did notice that if you slightly move the model so it's not in the center of the buildplate, I'm able to slice it. 
[image]

I hope that helps you to get unstuck. 🤞

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker