Ultimaker/Cura #13407: SD Card Error

FoxMulderX77, 2022-09-29:
### Application Version

5.1.1

### Platform

Windows 10

### Printer

Davinci XYZprinting PRO 1.0

### Reproduction steps

1. I've sliced some stl files in the very last release of CURA (the5.1.1)

### Actual results

2. Sending the model to the printer by using XYZprint software it ends with an SD card Error, then the printer stops.
3. This is not happening with the 5.0.0 release.

### Expected results

I used to print by this way with no issues ever before 5.1.1 version

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
https://drive.google.com/file/d/1sZvzQszQsifnSQEaOrZ3h1MuKlBpbHJD/view?usp=sharing

MariMakes, 2022-10-25:
Hey @FoxMulderX77 

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I'm not sure if what you are running into is an issue from Cura.
Are you still stuck? 
Did you try printing with other SD cards? 
Can you try and print something on the latest version (Cura 5.2.1)?

FoxMulderX77, 2022-10-25:
Well. During this time I found the cause for the SD Card error.
Actually the problem was the exporting of the geocode. 
[image]
By default the dialog window pops up with the Snapmaker G-code file format and I didn't notice before. Once I've used the traditional g-code the SD Card Error is gone.
[image]
Unfortunately, I cannot find a section in the preferences where I can change the exporting file format by default, so I could make this an automatic process. So it would be very useful include an option for this in the future.
Anyway, thanks for the answer and I hope this little change would be possible in the future.

MariMakes, 2022-10-26:
Hey @FoxMulderX77,

I don't see the option that you are seeing. 
[image]

Could it be that you have the snapmaker plugin installed?
https://marketplace.ultimaker.com/app/cura/plugins/macdylan/Snapmaker2Plugin

You can uninstall it by clicking this gear in the marketplace.
[image]

Does this solve your problem? 😊

FoxMulderX77, 2022-10-26:
Hello again. In fact, it was installed. I didn't noticed. I must have installed it by mistake. Once I've uninstalled the pluggin, the problem was gonne. Thanks for the help

MariMakes, 2022-10-27:
Yay! I'm happy to hear that. 😊 
I'll close this issue.

closed 2022-10-27 by MariMakes