Ultimaker/Cura #13408: filament change gcode unconsitant

antony3775, 2022-09-29:
### Application Version

5.1

### Platform

Windows 10

### Printer

Creality ender 3

### Reproduction steps

1. Created a post processing gcode script using cura filament change at a determinated layer height
2. Made sure i was setting the layer change at the starting layer of the new part/new print if printing one at the time.

### Actual results

sometimes it works sometimes it doesnt for example in the sport script i tell the printer to stop at layer 248 but instead it stops at layer 244 and goes back at printing the part that should be a single colour. sometimes triggers a layer too soon. from my understanding it stop at the very beginning of the selected layer. 

### Expected results

to stop at the correct layer for a filament change not a layer too soon or a serveal layers too soon

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3PRO_CAPRICORN_Cutting_3b_add_10mm__Option.zip](https://github.com/Ultimaker/Cura/files/9679610/CE3PRO_CAPRICORN_Cutting_3b_add_10mm__Option.zip)
layer changed one layer before twice and the last layer change was triggered 4 layers sooner than expected

SaschaUncia, 2022-09-30:
What a coincidence, I noticed I am having the same issue.

I create a multicolour layered print (0.25mm layers) and added pause at height 'modify gcode' at 3mm and 4mm. 

I initially expected this to pause after the 3mm layer was printed but it printed it in the new colour, so I changed it to 3.25mm and 4.25mm after, and got the same results.

Ok, so I figure, fine, it doesn't like me specifying the height in mm, so I switch to layers, and double check the layer number in cura. Layer 12 is that last layer that should print in black, so I add pause gcode at layer 13. Before printing and wasting more material, this time I open the .gcode file in notepad++ to verify that everything is correct. The pause gcode (m25) is at the end of layer 12, before layer 13 starts. Good. So I start the print, the printer print's right through layer 13 and pauses at the beginning layer 14.

Very frustrating

antony3775, 2022-09-30:
I printed an articulated monarch in red white and black and the layer change for the white happened a Layer before the targeted layer. Ruining the print. It was white so i just used a red permanent marker but its a failed print but i tried to save it. Its very frustrating indeed i thought it was my mistake maybe it wasnt

SaschaUncia, 2022-09-30:
I'm designing a red panda toothpaste tube squeezer  :3

[image]

SaschaUncia, 2022-09-30:
Oh and here's my gcode if it helps. The first pause was at layer 13, it added the pause code to the end of the layer 12 code-block but didn't pause until after 13 was done. Trying to specify a pause at layer 13 by setting a mm height instead causes the pause to happen earlier, both 3mm and 3.25mm resulted in layer 12 printing in the new colour (obscuring areas meant to be unmasked), very strange. 0.25mm layer height in all cases. 

https://drive.google.com/file/d/1QA2-2_DphS7VlyO95bsHZBW-j-nOCJnY/view?usp=sharing

antony3775, 2022-09-30:
Pretty cool

GregValiant, 2022-09-30:
Thanks for the report.
To be clear...
The layer numbering in a Gcode file is Base0.  The layer numbering in the Cura preview is Base1.
The filament change will take place at the start of the Layer # you insert in the dialog box.  If you insert layer 10 in the dialog then the pause will be at the start of the tenth layer which in the gcode file is Layer:9.

@antony3775 You are printing multiple models in "One at a Time" mode.  Because of that, the Cura preview is showing the total number of layers that will be printed (the Layer_Count).  When the gcode file is put together, the bottom of each individual model is on it's own Layer:0 (the Layer_Number).  There are 5 ";LAYER:0" lines in that gcode file.  The highest "Layer_Number" in the file is Layer:115 while the "Layer_Count" is 522.
So all your filament changes don't work because you are asking for numbers that don't exist in the file.
Someone from the Cura team will take a look.  Meantime, if you want to print that nut a different color then you will have to print it separately.

@SaschaUncia Your situation is different because you are using Pause at Height and not Filament Change.  If you are trying to Pause "at a height", and if there are Z-hops in the file, then the pause will occur at a z-hop rather than at the working height that you expected.  That is a known bug.
If you are trying to pause 1mm apart from another pause then there is a different issue.  The first Pause at say 5mm makes a 1mm move up to clear the print and so when the second Pause at Height looks for 6mm it sees it in the first pause at height script and so it is wrong.  This isn't really a bug but an unavoidable coincidence.
The workaround for that is to put the 6mm height change first in the list of post-processing scripts.  Cura will go through and put in the 6mm change and then go through again and put in the 5mm change and so the confusion doesn't happen PROVIDED there are no z-hops in the file.
If you feel strongly that this is a bug then you need to put in another report as it would be with Pause At Height rather than Filament Change.

SaschaUncia, 2022-09-30:
Done, thanks.

antony3775, 2022-09-30:
Alrigt but is there a way around this then other than filament change?

GregValiant, 2022-09-30:
Filament Change will work.  It's a single line command that passes parameters to the firmware.  The firmware then does the rest.  You will need to go into the gcode file (any text editor will open it) and make a couple of changes.
Each part on your build plate will start with a skirt.  So each print runs from ";TYPE:SKIRT" to the next ";TYPE:SKIRT".  That is a key to this.

Slice the file and put a single Filament Change at say Layer:10.  Save the gcode file.
Open the gcode file in a text editor and search for M600.
"Cut" the M600 line out of the gcode.
Search for a part name.  In your case "_3D" would be enough to find the nut.
From the " ;MESH:CAPRICORN_Cutting_3d.stl" line - Scroll upward to the ";TYPE-SKIRT" line.
Paste the M600 line just after ";TYPE:SKIRT".
Now search for the next ";TYPE:SKIRT".
Paste the M600 line just after TYPE:SKIRT.

So your print starts with Red Filament.  When it gets to the Skirt of part "xxxx_3D.STL" there is a filament change and you go to White Filament.  When "_3D" finsihes the nozzle will move to the next part and just before that skirt starts there will be another filament change where you would go back to Red Filament.  You end up with something like this and you would start with one color, change to another color for a specific part, and then change again when that part ends.

;LAYER_COUNT:18
;LAYER:0
G0 F7200 X188.139 Y112.338 Z0.4
G0 X187.276 Y115.689
;TYPE:SKIRT
M600 E30.00 U300.00 X0.00 Y0.00 ; Generated by FilamentChange plugin
>>>>>_a bunch of code for the skirt and model_<<<<<<<<<<
;LAYER_COUNT:23
;LAYER:0
G0 F7200 X23.45 Y100.708 Z0.4
G0 X18.955 Y104.952
;TYPE:SKIRT
M600 E30.00 U300.00 X0.00 Y0.00 ; Generated by FilamentChange plugin

antony3775, 2022-09-30:
I started a printing a another stl and i did what you suggested. Moved the m600 line after the skirt line. Will report back in 3 hours