Ultimaker/Cura #13409: Slicing wont work.

standan3126, 2022-09-29:
### Application Version

5.1.0

### Platform

windows 11

### Printer

Anycubic Max

### Reproduction steps

1. Put in my settings for my print.
2. Clicked slice and it said there was a bug to report.

### Actual results

it sents me here to report it.

### Expected results

I should have G code for my printer.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

cannot upload the files as they are too big

[Mary Parker Follett.pptx](https://github.com/Ultimaker/Cura/files/9679684/Mary.Parker.Follett.pptx)

standan3126, 2022-09-29:
it wouldnt let me upload my print file so i picked a school file to attach so i could file the report

MariMakes, 2022-10-25:
Hey @standan3126,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Thank you for this presentation, but I cannot troubleshoot your issue if I don't have your project file. 
Do you still have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2022-11-08 by github-actions[bot]