Ultimaker/Cura #13410: (5.1.1) Computer fully freezes when slicing with Wire Printing turned on

ruvilonix, 2022-09-30:
### Application Version

5.1.1

### Platform

Ubuntu 22.04

### Printer

Ender 3

### Reproduction steps

1. I imported one of any number of models.
2. I enabled Wire Printing.
3. I pressed Slice.

### Actual results

The progress bar begins to move. After a few seconds, the mouse begins to lag. The computer fan gets louder and louder and the mouse lag gets worse, until I can not move the cursor at all. 

The progress bar gets stuck at just under halfway. I can not open the terminal with any keyboard shortcuts. 

I rebooted the computer.

I tried several times, then did a full disk erase and reinstall of Ubuntu 22.04, but the problem persists.

### Expected results

I expected it to slice without freezing. No other setting that I've changed so far has caused a freeze.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Project file: [cup_holder_for_kayak.zip](https://github.com/Ultimaker/Cura/files/9679773/cup_holder_for_kayak.zip)

[cura.log](https://github.com/Ultimaker/Cura/files/9679774/cura.log)

MariMakes, 2022-10-25:
Hey @ruvilonix 

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I took a look at your file and it's a curious case 🤔 
I was able to slice your model easily as long as the X did not surpass 50 mm.
[image]

I'll bring it up to the team to take a look at.

We don't have a lot of users using this experimetnal setting. 
Can you share what you use it for?

MariMakes, 2022-10-26:
Hey @ruvilonix,

I brought it up to the team, but this is an experimental setting. To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

I hope you understand 🙏  You can still print this model if you resize it to be smaller.

ruvilonix, 2022-10-26:
@MariMakes No problem, thanks for looking at it.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker