Ultimaker/Cura #13411: Cura won't slice model

mrsilbarita, 2022-09-30:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

I can't give specific steps, but it generally happens when slicing particularly large models (40hs or more). Resetting the program has worked but doesn't always.

Also, weirdly enough it appears to happen on a threshold: if I make the model 50% size it might work and also on 80% but not 100%.

### Actual results

Slicing gets stuck here forever.

[image]

### Expected results

Slicing proceeds as normal.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[AET5_Africa 1.zip](https://github.com/Ultimaker/Cura/files/9680138/AET5_Africa.1.zip)

mrsilbarita, 2022-09-30:
It was brought to my attention that this happens when you have two monitors, and indeed turning one of them off solved the issue.

GregValiant, 2022-09-30:
Thanks for the report.
There have been issues with certain configurations of multiple monitors.  That aside...
The warning "The model is not watertight and may not print correctly" came right up for me.  In some cases models that aren't watertight won't slice either.  In the case of that model, it has a lot of errors.  Here is the report:
--> 3 Naked edges (?)
--> 1 Planar holes (?)
--> 0 Non-planar holes (?)
--> 338 Non-manifold edges (?)
--> 651 Inverted faces (?)
--> 8 Degenerate faces (?)
--> 676 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 77068ms ------
-> Vertex count changed from 161387 to 164953 (+3566)
-> Triangle count changed from 322733 to 329898 (+7165)
An app like MS 3D Builder or Mesh Mixer could also repair the model.
There is also an often reported bug in 5.1 that has a workaround.  Move the model on the build plate and/or rotate the model.

Someone from the Cura team will look at this.

mrsilbarita, 2022-09-30:
I am using multiple monitors actually, so I'll look into that.

I know it's a messy model, but it's about the right amount of jank that I usually get to work.

I also tried rotating it and it sliced this time, so there's that.

Thanks a lot!

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker