Ultimaker/Cura #13415: "api-ms-win-core-path-l1-1-0.dll is missing"

The-Watching-Tiger, 2022-09-30:
### Application Version

5.1.1

### Platform

Windows 7 Ultimate

### Printer

Creality Ender 3v 2

### Reproduction steps

I downloaded and installed the Ultimaker Cura 5.1.1 after launching the Software, an error occurs saying: "api-ms-win-core-path-l1-1-0.dll is missing"

### Actual results

I think maybe i should look for a patch for Ultimaker Cura 5.1.1
I tried to uninstall and reinstall the software and i tried the Beta version of cura too.

### Expected results

Maybe some files are dameged or being removed from my system, i could use to install the cura on another computer but i would apriciate your help anyway.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

vikmaks1, 2022-09-30:
It is due to incompatible operating system, newer Python which is used for the current Cura versions does not support Windows 7. So you are either stuck with older versions or use a computer that is upgradable to Windows 10.

MariMakes, 2022-09-30:
Hey all!

@vikmaks1 is right! Windows 7 is indeed not on our supported Operating System list.
For that, you'll have to go back to Cura 4.10.

More about it here:
https://support.ultimaker.com/hc/en-us/articles/360011889579

I'll close the issue since it's expected behavior 
Happy Printing 😄

closed 2022-09-30 by MariMakes