Ultimaker/Cura #13417: Cura Drops LSD On Preview Render in MacOS

EasyG0ing1, 2022-09-30:
### Application Version

5.1.0

### Platform

MacOS 12.5.1

### Printer

Anycubic Mega X

### Reproduction steps

Second time this has happened to me and the only way I can fix it is by completely wiping the app and related folders then re-install Cura, but for some reason, when it renders previews, it looks like it's on acid ...

[see screen shot](https://i.imgur.com/fkBvz7I.png)

### Actual results

[see screen shot](https://i.imgur.com/fkBvz7I.png)

### Expected results

Not that

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-10-02:
Thanks for the report.
This is one of the cockroach bugs of Cura.  It's hard to duplicate and consequently has been impossible to get rid of.
After saving the gcode, open the gcode file in Cura.  I believe you will find that all that peripheral stuff is not present in the gcode.

Is this happening with all your models?  If so, then please zip a project file ("File | Save Project") and the Cura.log file ("Help | Show Configuration Folder") and post the zip file here.  If you could also post some information about your video sub-system that could be helpful.  Someone from the Cura team will take a look.

On a less serious note, it's possible you have installed the Timothy Leary version of Cura.  It uses Peter Maxx rendering.  You can see the effect in this slice of three older model benchies.
[image]

EasyG0ing1, 2022-11-09:
@GregValiant It will take some time before I can provide the additional information requested, since I reset my install after creating this issue. It can take sometimes months before the problem shows up again, so the best I can do is get you what you're asking for the next time it happens.

GregValiant, 2022-11-09:
That's fine.  Like I said, the bug has been around a while and it's tough to duplicate.  It could be related to your specific hardware, or a glitch in the Cura code, or your video system driver doesn't play well with Cura's OpenGL requirements.  Nobody knows for sure.
When it happens again try to get good documentation.  The more information the better for debugging.

I'm going to close this report as there just isn't much we can do at this point.  If it happens again you can open another report.

closed 2022-11-09 by GregValiant

EasyG0ing1, 2022-11-17:
@GregValiant - If it matters, I'm running on a 2019 MacBook Pro ... On my last MBP I recall being able to choose which graphics card was the active card but I can't seem to find that option on this one ... it would be an interesting test to see if it had any affect on the problem.

[image]

EasyG0ing1, 2022-11-17:
@GregValiant - The other interesting factor about this issue, is that it takes time (usage of the software) before it happens, and wiping out the install and re-installing from scratch fixes it... which begs the question, what could possibly be changing in the softwares files themselves that would have any impact at all on the rendering of preview images?

GregValiant, 2024-11-16:
I found another reference to this bug #10387