Ultimaker/Cura #13420: Vase mode now creates extra/additional walls on alternating layers in the beta

PhilBaz, 2022-10-01:
### Application Version

5.2 Beta

### Platform

10

### Printer

Voron 2.4

### Reproduction steps

Vase mode now creates extra/additional walls on alternating layers in the beta....

[image]
[image]

### Actual results

Extra walls

### Expected results

requested number

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[spiralize creates extra walls.zip](https://github.com/Ultimaker/Cura/files/9689228/spiralize.creates.extra.walls.zip)

MariMakes, 2022-10-03:
Hey @PhilBaz,

Thanks for the report 👍 

We restored the spiralize behavior as it was in 4.13 because we've gotten requests from users, like yourself, to restore it. 
This is the behavior in 4.13 - 5.1 - 5.2 BETA
[image]

So what you are seeing is intended behavior. 
These alternating layers proved to create a sturdy base.

Do you have any examples that actually impact your printed part quality negatively? 
Then we can reconsider the implementation.

closed 2022-10-03 by MariMakes

PhilBaz, 2022-10-03:
@MariMakes ahhh, okay...

No I don't think it will have a negative impact on quality. Its just not expected since I didn't enabled it through any settings. Since you don't request it, it feels like a bug. 

Also, since vase mode only contacts the outermost wall, and skin generally dose a fine job of making a sturdy base, it just feels like you lose control of the outcome. If I want more walls I can always add them. But, if I don't want the extra, I cant remove them.