Ultimaker/Cura #13422: Slicing failed

olsfors, 2022-10-01:
### Application Version

5.2.0 beta

### Platform

Linux Mint 21

### Printer

Ultimaker S5

### Reproduction steps

Slicing with Engineering Fine 0.1 mm, 20% fill, Tough PLA

### Actual results

Got message: Unable to slice.

### Expected results

Slicing finished

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9690222/cura.log)
[top-baro.zip](https://github.com/Ultimaker/Cura/files/9690226/top-baro.zip)

MariMakes, 2022-10-03:
Thanks for the report @olsfors 👍 
And thanks for reporting the duplicates @Piezoid ❤️ 

I'll bring them up to the team. 🤞

MariMakes, 2022-10-03:
We're working on a solution to make it work again. 💪
For internal reference CURA-9711

Thanks for the reports ❤️

nallath, 2022-10-06:
I've uploaded a new version of the beta. Enjoy!

commit ca90ac77 (2022-10-05): Fix adding all shared libs managed by Conan to pyinstaller

Closes #13422

Contributes to CURA-9711
closed 2022-10-10 by jellespijker