Ultimaker/Cura #13426: Cannot slice on Cura 5.2.0-beta

uski, 2022-10-02:
### Application Version

5.2.0-beta

### Platform

Linux

### Printer

Creality Ender 3 Pro

### Reproduction steps

1. Import a STL
2. Try to slice it

### Actual results

Error message : Slicing failed with an unexpected error

### Expected results

Proper slicing

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Unfortunately there is another bug that prevents me from saving a project file (the open and save dialogs do not show, and I can only open an STL by doing a drag&drop into Cura). I have however attached the STL.
[curabug.zip](https://github.com/Ultimaker/Cura/files/9691843/curabug.zip)

uski, 2022-10-02:
Just tried slicing the same file on Cura 5.1.1 and it worked without any issue.

Piezoid, 2022-10-02:
I believe this is a duplicate of #13422, an issue with AppImage bundling
```
2022-10-01 22:37:57,976 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [115]: [Backend] /tmp/.mount_Ultima6VI939/CuraEngine: error while loading shared libraries: libspdlog.so.1: cannot open shared object file: No such file or directory
```

MariMakes, 2022-10-03:
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13422
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13422

closed 2022-10-03 by MariMakes