Ultimaker/Cura #13428: Raft generation problem in 5.2.0-beta

hoblin, 2022-10-02:
### Application Version

5.2.0-beta

### Platform

Mac (M1)

### Printer

Creality Ender 3 Max

### Reproduction steps

1. Enable raft
2. Slice model
3. Check resultthe 

### Actual results

Raft generates under full square under the model

### Expected results

Raft follows the model shape

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Two project files were uploaded. 
1. [Cura-5.1.0_CE3MAX_iPhone_AppleWatch_Stand.3mf.zip](https://github.com/Ultimaker/Cura/files/9692086/Cura-5.1.0_CE3MAX_iPhone_AppleWatch_Stand.3mf.zip) version with proper raft generation
2. [Cura-5.2.0-beta_CE3MAX_iPhone_AppleWatch_Stand.3mf.zip](https://github.com/Ultimaker/Cura/files/9692087/Cura-5.2.0-beta_CE3MAX_iPhone_AppleWatch_Stand.3mf.zip) version with a too big raft

Cura 5.1.0:
[image]
Cura 5.2.0-beta:
[image]

MariMakes, 2022-10-03:
Hey @hoblin,

Welcome to the Ultimaker Cura Github 🚀 

This behavior looks like the Remove Raft Inside Corners setting.
[image]
If you uncheck it you see the original behavior.

The fact that it's turned on by default seems to be faulty behavior. 
I'll let the team know.

Good find! 👍

MariMakes, 2022-10-03:
I discussed it with the team.
It was a mistake on our side. 

We resolved it in https://github.com/Ultimaker/Cura/commit/b33cff33f5f18f7be23e6c0a86a302ea887a517c and it should be resolved in the stable version of 5.2 so I'll close this issue. 

Thanks again for the report ❤️

closed 2022-10-03 by MariMakes