Ultimaker/Cura #13429: Splash and Main Screen don't remember position or size on multi screen computers

MarkALynch, 2022-10-02:
### Application Version

5.2.0

### Platform

Windows 10

### Printer

Any Creality 

### Reproduction steps

Open Cura 5.1 or 5.2 and the splash is always on the left (of 3) screen. When the main window opens it's always the default size, so it clips the lower buttons. When I resize it to fill the screen (not maximized) and close, and then open, it returns to the default size it was coded for.

### Actual results

Splash and main screen are in default positions / size.

### Expected results

Splash doesn't really matter what screen it appears on, but the main screen should remember size and position on the screen it was last run on.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Since this started in version 5, I've uploaded log for 5.1 and 5.2
[cura.log](https://github.com/Ultimaker/Cura/files/9692288/cura.log)
[cura.log](https://github.com/Ultimaker/Cura/files/9692291/cura.log)
[image]

MariMakes, 2022-11-01:
Hey @MarkALynch,

Sorry, it took us a while to get back to you 😞 
[When I read this it reminded me of this workaround I found on the community.](https://community.ultimaker.com/topic/41458-ultimaker-cura-510-window-scaling-problem/#comment-311838) 

1. Pull up Cura 5.1.0 on the Start menu
2. Right click on it and choose Open File Location
3. Right click on the shortcut and choose Properties
4. Click on the Compatibility tab
5. Click Change high DPI settings
6. Click Override high DPI scaling behavior and choose System (Enhanced) from the drop down.

Can you let me know when you try this, are you still encountering the resizing problem?

MarkALynch, 2022-11-01:
This made no difference.

My screens are laid out as follows   1  3  2

My screens are sized at 1920x1080 each.

When I open Cura, the splash always appears on screen 1, centered.
When I position Cura main screen on Screen 3, resize to about 90% of the screen, close it, it always opens at the default size, not what I set/stretched.
When I position Cura main screen on Screen 1, resize to about 90% of the screen, close it, it always opens at the remembered (what I set) size, properly
When I position Cura main screen on Screen 2, resize to about 90% of the screen, close it, it opens on screen 3 at the default size, not what I set/stretched.

So, as long as Cura main screen is used on Screen 1, it will remember screen position and size, but on Screen 2 or 3 it doesn’t remember position or size.

This started with Version 5 Beta, version 4.x all worked across any screen.

Thanks,
Mark Lynch

W0EB, 2022-11-01:
> From "MariMakes" ***@***.***>
> To "Ultimaker/Cura" ***@***.***>
> Cc "W0EB" ***@***.***>; "Comment" ***@***.***>
> Date 11/1/2022 12:05:15 PM

Hey Mark,
I had tried that long before I even posted the problem.  It still does 
the same thing.  Running the lastest 64bit version of Windows 10

<https://community.ultimaker.com/topic/41458-ultimaker-cura-510-window-scaling-problem/#comment-311838>

W0EB, 2022-11-01:
I tried it again, followed your directions EXACTLY and Cura 5.2.1 still 
has the exact same problem.  I am using the latest updates on Windows 10 
64bit if that makes a difference.

Jim Sheldon

<https://community.ultimaker.com/topic/41458-ultimaker-cura-510-window-scaling-problem/#comment-311838>

aerobaticant, 2022-11-21:
I'm having exactly the same problem. Cura only remembers the window position if it is used on the monitor where the splash appears, in my case the right hand monitor in a three monitor setup. Also Windows 10, 64-bit.

astrine, 2023-07-10:
I have exactly same problem in version 5.3.0. Is there any updates?

W0EB, 2023-07-11:
Astine,
That problem seems to have been resolved, but there is another much 
larger problem.  I use windows 10 professional, and the latest version 
doesn't slice properly at all for my CR10 V2 printers.  All holes are 
filled regardless of setting, the print time (should be over an hour) 
shows 11 minutes and the print turns up garbage.

That was a problem on the previous version as well.

I have reverted to using version 5.2.0-beta which is the only version 
that works properly on my computer and with my printers.

Aubrey

From "Mariska" ***@***.***>

Date 7/11/2023 10:28:39 AM

@astrine <https://github.com/astrine>, can you confirm if it's resolved 
in the latest cura version?
You can download it here: 
https://github.com/Ultimaker/Cura/releases/tag/5.4.0

😮 Since we have a solution in place, would you still like to keep this 
issue open or do you agree that we can close it?

MariMakes, 2023-07-17:
Hey @W0EB,

That sounds like something different than your original issue 🤔 
It would be easier to resolve if you create a new bug report for it here: https://github.com/Ultimaker/Cura/issues/new/choose

But at first glance it seems to be related to one of the settings in Cura. 
Can you check the following settings for me:
- Remove All Holes;  Remove the holes in each layer and keep only the outside shape. This will ignore any invisible internal geometry. However, it also ignores layer holes which can be viewed from above or below.
- Extensive Stitching, Extensive stitching tries to stitch up open holes in the mesh by closing the hole with touching polygons. This option can introduce a lot of processing time.
- Spiralize Outer Contour, Spiralize smooths out the Z move of the outer edge. This will create a steady Z increase over the whole print. This feature turns a solid model into a single walled print with a solid bottom. This feature should only be enabled when each layer only contains a single part

Since your issue has been resolved. I'll close this ticket, but if you still run into another problem, please open a new one so we can keep track of it. Thank you 😊

closed 2023-07-17 by MariMakes