Ultimaker/Cura #13430: Reduce the allotted space for hotend movement when printing "one at a time"

middlechamber, 2022-10-02:
### Is your feature request related to a problem?

Not so much a problem but efficiency of usable print space

### Describe the solution you'd like

 I print many objects that are small to miniature (some as small as 5-6mm X & Y, 1.8mm Z) and usually print using the "one at a time" mode to reduce stringing & possible failure.  But at the same time, I can't fill the build surface even though what I'm printing is under 6mm Z and the X&Y measurements would allow me to place 12-18 or more on the build surface printing "all at once" with a separation of upwards of 10mm between the pieces, because of the area allotted for the movement not hitting a preprinted item or the gantry in the way. 
Is there any way when printing items that have Z height under the height of the nozzle exposure (6mm), to reduce that "greyed out area" as it is when you print "all at once" ? 
Or could that be a possibility in any upgrade?

### Describe alternatives you've considered

May be adding a sub box mode under (print one at a time) that has height restrictions to whatever the nominal nozzle height is that could be set in the settings? or at least a minimum height allowed.  

### Affected users and/or printers

I think any experience person printing items as I do would appreciate that feature

GregValiant, 2022-10-04:
Thanks for the request.
The Printhead Settings are available in the Machine Settings.  You can alter the numbers and make the "crash" shadow smaller or larger.
The numbers are measured from the centerline of the nozzle and so X Min and Y Min will both be negative numbers.  The Gantry Height is (generally) from the nozzle tip to the underside of the X beam on a cartesian printer.  On my Ender you can see that it is 25mm.
I would suggest that you add a second printer in Cura, give it a different name, and make your changes to the PrintHead settings (and possibly Start and End gcodes) of the second printer.
When you have a print with numerous short pieces and you aren't worried about whacking a completed model you pick the second printer.  Your main printer would still have the "no crash" safeties in place.
[image]

middlechamber, 2022-10-04:
Hi,     Thanks for the replay, I wish I understood part of what you expressed. I'm old & never had any formal IT training, I understand naming the printer as a second unit, but I don't do gcode, Pi, or SD card, I have always used the USB connection to my PC.  *When I first got the printer over 4 years ago, I wasn't on any FB self help site or watched any videos. The SD card was snapped in half, so I downloaded what I thought I needed & went from there, and have been printing successfully as a business.    I've never had a problem with the printhead or gantry hitting a printed item when I print multiples "one at a time" they're always under 6mm in height Z, the gantry being even higher always raises then positions itself to the next start point so there is no concern for that. I have no clue what printhead settings to install if that's the solution as you suggested, and don't want the nozzle crashing into the glass build surface.  Any suggestions for items no higher than 6mm?

GregValiant, 2022-10-04:
The things I talked about are adjustments you can make in Cura.  If you go to Settings / Printers / Manage Printers and then click on Machine Settings you will see the window I showed above.  On the right side make the numbers like this:
X min = -5
Y Min = -5
X Max = 5
Y Max = 5
That will make a 10 x 10 shadow on the build plate and you could get your models much closer together.
You should also change the Skirt Line Count to 1 and the Skirt Distance to 2.0.  That way the bed adhesion won't take up as much room.
Here I've made those changes and you can see that I've gotten 16 pieces on the 230 x 230 build plate.
(And I'll bet ya dollars to donuts that I'm older than you.)

[image]

middlechamber, 2022-10-04:
Hi,     I'm not too sure about the age thing.... 70 & counting.Not that I'm totally incapable to figure out when/if I had time & a few adult beverages, but we started Christmas orders in August & will be at it till 2 weeks before the holiday. If I loused up just 1 machine my butt would be in a sling. My wife carries a mean whip & knows how to use it...Lol! but seriously, we are in the middle of Christmas production for her Art Business. I appreciate the numbers when the printers stop tonight, I'll have at it & test. Thank you,                  Paull.

GregValiant, 2022-10-04:
I've got you by 2 and I ain't counting any more.
It looks like the silicone sock on my nozzle is about 10mm in diameter so if yours is about the same then you shouldn't get any crashes.