Ultimaker/Cura #13431: Bridges print inner walls in midair

Dasky14, 2022-10-02:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Ender 3v2

### Reproduction steps

1. Added bridge test calibration model (from "part for calibration" plugin)
2. Set walls to 3+
3. Slice

### Actual results

Bridges print inner walls completely in midair, resulting in the middle of the bridge failing entirely.

[image]

### Expected results

With only 1 wall, the bridge is printed properly with no walls in midair.
Sadly using a single wall for the entire model is often not an option, since this results in poor structural integrity and a ton of quality issues.

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project file:
[Bridge_issue.zip](https://github.com/Ultimaker/Cura/files/9692488/Bridge_issue.zip)

Dasky14, 2022-10-25:
Maybe technically not a bug, but definitely unwanted/unintended behaviour.

While this solution works for this particular model, if I had a model with thicker parts to it and wanted 1.8mm walls on the rest of the model (to avoid infill leaking through, or for structural integrity reasons for example), then these bridges would fail and there is no way for me to just change this particular part of the model to print with fewer/thinner walls.

Printing with thinner walls is simply not always an option.

Jerooney, 2022-10-26:
Hi @Dasky14 ,

I understand your frustration and failed print, but what @MariMakes says is true:
'Top and bottom layers are generated based on the perimeter of the print. From this perimeter, it will create paths outside to inside, for the number of 'walls' set by the user.'

This means, we actually define here what to do. And there's the catch: We have this quite long bridge with simply very thin pillars. Increasing the pillar size will allow the perimeter to be wider on top of them, instead of floating. It's a 'edge case scenario'.

**Fix:** Please use the 'extra infill wall count' to improve strength on parts that have a bigger perimeter than your 'thin parts' where bridges are in place.

My advice on your settings:
Wall count =2
Extra infill wall count = 2

This will give 4 walls in the model, where there is place to build them. Let me know if this helps your case.

Dasky14, 2022-10-27:
Well, I suppose that's good enough. I still think that this particular edge case could have a better solution in the slicer, but this is probably not the place to debate stuff like that.

I'll close the issue as resolved.

closed 2022-10-27 by Dasky14