Ultimaker/Cura #13433: Cura 5.1.1 starts and closes immediately

kn4ycd, 2022-10-02:
### Application Version

5.1.1

### Platform

OS X Catalina

### Printer

FL Sun Super Racer

### Reproduction steps

Upgraded from 5.01 to 5.1.1.  5.01 was working just fine.

### Actual results

after 5.1.1 installed, i try to start Cura, and it never starts (that I can see).  It is not in the process table, it just isn't running at all.  

### Expected results

Cura should work

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No log files, as the app is not creating the folder path for cura 5.1.1 at all.  the latest files i have are from several days ago, for 5.01

kevinhrim, 2022-10-12:
I'm also running on Catalina (10.15.7). with same behavior. 

From the console log:
Oct 12 17:21:48 server1 com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x10000003.Ultimaker-Cura[3555]): Service exited with abnormal code: 255

Mac Analytics:
17:19:34.334908-0400	syspolicyd	assessment granted for Ultimaker-Cura-5.1.1-mac (1).dmg by Notarized Developer ID
com.apple.message.domain: com.apple.security.assessment.outcome2
com.apple.message.signature2: bundle:UNBUNDLED
com.apple.message.signature3: Ultimaker-Cura-5.1.1-mac (1).dmg
com.apple.message.signature5: UNKNOWN
com.apple.message.signature4: 3
com.apple.message.signature: granted:Notarized Developer ID
SenderMachUUID: 1BF18A31-77EA-3C23-B1AC-920F9A9DC507

kevinhrim, 2022-10-12:
One additional comment.  Other versions prompt about the network security setting (allow inbound network connections)  5.1.1 did not prompt for that permission.

MariMakes, 2022-11-08:
Hey @kevinhrim,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

Cura 5.1.1 and up are unfortunately not supported on Mac OS 10.15 like we mention in the release notes:
https://support.ultimaker.com/hc/en-us/articles/360011889579

If upgrading is not an option for you, you'll have to stick with Cura 5.1.0 as mentioned in the article.

To me, it doesn't seem like a bug in Ultimaker Cura.
Apologies for any inconvenience caused by our delayed response. 
I'll be closing this issue. I hope you understand.