Ultimaker/Cura #13434: (Windows 10) Fatal Error: sqlite3.DatabaseError: file is not a database

KnottyDyes, 2022-10-02:
### Application Version

5.1.1

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

My system had been crashing due to a failing GPU.  It had crashed enough to cause some programs to have issues with their configurations.  Cura is apparently one of these apps.  After one of these crashes and subsequent reboots, Cura refuses to launch.  I have uninstalled and reinstalled Cura multiple times, used Revo uninstaller to rip out as much as possible, yet still it crashes with the same errors.  

### Actual results

`Traceback (most recent call last):
  File "UM\Settings\ContainerRegistry.py", line 469, in loadAllMetadata
    db_last_modified_time = self._getProfileModificationTime(container_id, cursor)
  File "UM\Settings\ContainerRegistry.py", line 415, in _getProfileModificationTime
    db_cursor.execute("select id, last_modified from containers where id = ?", (container_id, ))
sqlite3.DatabaseError: file is not a database

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "UM\Settings\ContainerRegistry.py", line 469, in loadAllMetadata
    db_last_modified_time = self._getProfileModificationTime(container_id, cursor)
  File "UM\Settings\ContainerRegistry.py", line 415, in _getProfileModificationTime
    db_cursor.execute("select id, last_modified from containers where id = ?", (container_id, ))
sqlite3.DatabaseError: file is not a database

During handling of the above exception, another exception occurred:

---- continue to loop ----
`

`Thread 0x00007ff8 (most recent call first):
  File "UM\Backend\Backend.py", line 165 in _storeStderrToLogThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00006948 (most recent call first):
  File "UM\Backend\Backend.py", line 153 in _storeOutputToLogThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00007a94 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000022f8 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001bc0 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000042d8 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00004af0 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000023f0 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000074b0 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00003df0 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00008afc (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001198 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000063e8 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00003a14 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00000f28 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000025b0 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000028d8 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000848c (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000269c (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000062c4 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00007f18 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap
`

### Expected results

Launch Cura and slice things.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9692822/cura.log)

fieldOfView, 2022-10-04:
If you remove the file `containers.db` from `C:\Users\[your username]\AppData\Local\cura\5.1\cache`, it should be recreated on the next launch of Cura.

Please let us know if that helped.

fieldOfView, 2022-10-04:
Important addition: note that the file is in the `local` AppData, not the regular `roaming` AppData folder.

closed 2022-10-19 by github-actions[bot]