Ultimaker/Cura #13436: tronxy x5sa-400 pro profile won't slice

WOWElectronics, 2022-10-02:
### Application Version

5.2.0-beta-linux

### Platform

debian 11 uptodate

### Printer

tronxy-x5sa-400 pro

### Reproduction steps

downloaded beta 5.2.0
chmod +x
softlink to cura
ran cura, got 5.2.0 signon.
it loaded the tronxy printer profile ok
load an .stl from OpenSCAD the worked with 5.1.0
laid it down on the plate, its only 3 mm thick
hit slice, look at busy pointer till the cows come in balling because their bag is heavy.
hit cancel after a couple minutes, get request to make bug report. You are looking at it.

 I'll go fix the link to run 5.1.0 again, I've got work to do.

### Actual results

slice, no progress

### Expected results

expected it to slice in 2 or 3 seconds, its a carriage plate part for a 3d printer. I'll make it out of PETG+CF after I locate about 6 more boltholes.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

2020_carriage.stl
whatdoyamean no support for that file type.

MariMakes, 2022-10-04:
Hey @WOWElectronics,

Thanks for the report. 👍 
We are aware that all users using an AppImage of the 5.2 Beta are encountering problems slicing.
Our team is working hard on a solution. Apologies for any inconvenience 😞 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13422

closed 2022-10-04 by MariMakes

nallath, 2022-10-06:
I've uploaded a new version of the beta. Enjoy!

WOWElectronics, 2022-12-16:
It turns out that the marlin in the tronxy-400-pro is buggier than 10 day old road kill in late july. The only place it will eject hot plastic is about 4mm above the bed, so It has yet to print anything for my $700 investment. So I have embarqed on a program to put an octopus-1.1 board in it with a 650 watt 48 volt motor supply, with a CF X bar to lighten the load, and stepperonline stepper/servo's to replace the home losing bblb motors in it. Progress is slow, in part because its hard to find a printer under $5000 that can actually survive a steady diet of PETG for more than a week. Makers lie like a rug. If and when, I get to the tronxy, its about 3rd in line for repairs ATM as I just blew up the motherboard in a twotrees saphire5-plus, the printer I am using to make parts to do the octopus hi voltage conversion to an Ender 5 plus. 

In the meantime cura 5.2's only problem is asking for top skin, it will freeze up the hotend about 99% of the time even with retractions below 3mm, and reboring the hotend on a saphire 5 plus is a major project. You've got to take it apart, get the heat break out, take it and the nozzle to the gas cookstove and burn it clean.  And it looks like I'm going to have to buy an ultrasonic cleaner to get all the ash out of them reliably.

3d printers are nice, when they work, but are a black hole for money too.

Take care, stay warm and well all.