Ultimaker/Cura #13438: Slicing fails with specific expansion setting

orbitalr, 2022-10-02:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Custom

### Reproduction steps

Slice and print with (some) horizontal expansion settings. 
 Horizontal expansion : 0.5
 Initial layer Horizontal expansion : -0.2
 Hole Horizontal expansion : 0.25

Change  Horizontal expansion to 0.25

Slice. 

 

### Actual results

Slicing failed. 

Tested:

 Horizontal expansion : 0 - OK
 Horizontal expansion : 0.1 - OK
 Horizontal expansion : 0.2 - Failed
 Horizontal expansion : 0.21 - OK
 Horizontal expansion : 0.22 - OK
 Horizontal expansion : 0.23 - OK
 Horizontal expansion : 0.24 - Faled
 Horizontal expansion : 0.26 - OK
 Horizontal expansion : 0.27 - Failed
 Horizontal expansion : 0.28 - Failed
 Horizontal expansion : 0.29 - Failed
 Horizontal expansion : 0.3 - OK

### Expected results

Successul slicing. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[SlicingBug20221002.zip](https://github.com/Ultimaker/Cura/files/9692973/SlicingBug20221002.zip)

orbitalr, 2022-10-03:
> Thanks for the report and project file.
> I am able to duplicate this behavior with your project file and using 5.2beta. Someone from the Cura team will look at this.
>
> Another problem is that the top of the model is at an angle. Staring with layer 68 the part no longer has a "hole" because the slices become horseshoe shaped (open on one side) rather than a closed shape.
> I set the Hole Horizontal Expansion to 1.0 for this.
> Here is the situation below layer 68. Hole Horizontal Expansion is being applied to layers 1 through 67.
> [image]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F64202104%2F193571475-bb0ec29c-0cda-4d99-a642-8715d777d596.png&data=05%7C01%7C%7C5af8e893076f4165062208daa539cffb%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638003964973383877%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=FBJL52T7PtbBWLf2w6MMmuSmrzHn6mu8rP7wbH9GKGc%3D&reserved=0>
>
> Here is layer 68. It can't be considered a "hole" anymore and so you would have a step in the inside surface of the print starting at this layer. You can see how much wider the slice has gotten because HHE cannot be applied from this point upward.
> [image]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F64202104%2F193571581-a6bcd59c-ccc7-44ae-abf1-c77acbeedd35.png&data=05%7C01%7C%7C5af8e893076f4165062208daa539cffb%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638003964973383877%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=BmH3T4EHd2cIUGqrdMXvYs7S9V7dB6or%2BnVADaWDEKs%3D&reserved=0>

Hi Greg,

Yes, that is an accurate assessment of the printed result.

I didn't mention the HHE behaviour as this could be "by design" rather than a bug. Although, realistically, a "hole" doesn't have to be only vertical.

My understanding of horizontal expansion correction is that the model becomes fatter, therefore voids smaller. If there is a better way of making the perimeter of a model larger while (or correcting) void dimensions please let me know where I can read about it.

Thanks.

GregValiant, 2022-10-03:
You could scale the model up by a fraction of a percentage.
There are a couple of new settings in the Material section that are for shrinkage.  You might get a different effect using them.

_The best thing to do is change the design in CAD so it prints at the size you want._

HHE isn't going to work like you want/expect because of the angle.  The upper part of the model only has a single outer wall that goes around the inside and continues around the outside back to a start point and so can't be considered as a hole.  The lower portion has an outer wall facing inside (the ID) and an outer wall facing outside (the OD) and is mathematically a hole.  Holes don't need to be round either.  Remember that there are no circles in FDM, just approximations made up of a lot of line segments.