Ultimaker/Cura #13439: multi-extruder support lock

Looping-39, 2022-10-02:
### Application Version

5.2.0 BETA

### Platform

windows 10

### Printer

CRX Pro

### Reproduction steps

I wanted to block supports on a part where there was no need for them, cutting was impossible.
The cause of the problem is that only my right extruder is enabled on my machine because my left extruder is down.

### Actual results

To solve the problem, it took having the 2 extruders activated in "prepare", I changed the support extruder in the print settings (even if the support option was disabled), and slicing was possible .

### Expected results

If both extruders are on, there is no slicing problem. If only one is activated and you want to put supports, slicing is impossible.
[supports cura.zip](https://github.com/Ultimaker/Cura/files/9693051/supports.cura.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I'm a beginner and I don't have a lot of money. Cura is really awesome. thank you for everything

GregValiant, 2022-10-03:
Thanks for the report.
The problem may be that when support blockers are brought in they are (by default) always assigned to extruder 1.  When you set your model to print with extruder 2 and then disable extruder 1, Cura notices that the support blockers are set to a disabled extruder and won't slice.
So select each Support Block on the Cura build plate, and on the left of the screen set each to the extruder you intend to use. That should fix the problem.  You will also want to check that the Bed Adhesion extruders are set to your enabled extruder.
The way it is working is expected behavior and so I don't see a bug here. I'll remove the tag and close this report.  If you think that is premature you can still respond here and I'll be notified.

closed 2022-10-03 by GregValiant

Looping-39, 2022-10-11:
sorry to have disturbed you for so little.  excellent continuation.  and
thanks for Cura.