Ultimaker/Cura #13440: Bad HTTP Link in application, goes to 404 page

themobydisk, 2022-10-02:
### Application Version

5.1.1

### Platform

Windows 10

### Printer

Ultimaker S3

### Reproduction steps

1. Launch Cura with an existing Ultimaker printer configured
2. Click Preferences - Configure Cura - Printers
3. Select a printer and click the "Connect via Network" button
4. Click the link at the bottom that says "If your printer is not listed, read the network printing troubleshooting guide."

### Actual results

A browser is opened pointing to a 404 page.  The URL is: https://ultimaker.com/en/cura/troubleshooting/network?utm_source=cura&utm_medium=software&utm_campaign=manage-network-printer

### Expected results

A browser is opened pointing to the network printing troubleshooting guide.  I suspect that the link should be: https://support.ultimaker.com/hc/en-us/articles/360012795419?utm_source=cura&utm_medium=software&utm_campaign=add-network-printer-error

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

There are 2 other places in the application that have good links to https://support.ultimaker.com/hc/en-us/articles/360012795419?utm_source=cura&utm_medium=software&utm_campaign=add-network-printer-error

1) Preferences - Configure Cura - Add New: Troubleshooting (top right).  
2) Preferences - Configure Cura - Add New - Add Printer by IP - (type in an IP) - Add - "Can't connect to your Ultimaker printer?" (bottom)

MariMakes, 2022-10-03:
Hey @themobydisk,

Thanks for the report 👍 
I can recreate your issue, this is definitely not intended. 
I'll let our team know to adjust the link. 

[image]

Great find 👍

MariMakes, 2022-10-03:
Hey @themobydisk,

I've informed our web team and they have published the fix online.
The link is working again. 🎉 

Thanks again for the report 👍

closed 2022-10-03 by MariMakes