Ultimaker/Cura #13441: z-seam is not smooth

passkal, 2022-10-02:
### Application Version

5.2.0

### Platform

windows 10

### Printer

Ender 5

### Reproduction steps

In all versions of CURA 5.x, the Z-seam on the round surface (of my model) turns out to be torn, only for this reason I use only the CURA 4.13.1 version

[image]

### Actual results

In all versions of CURA 5.x, the Z-seam on the round surface (of my model) turns out to be torn,

### Expected results

Please pay attention to this problem

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

passkal, 2022-10-02:
Another angle of the model
[image]

GregValiant, 2022-10-03:
Thanks for the report.
This is a duplicate of #11877 (another report is #12100).  I'll mark it as a duplicate and close your report.  You can follow progress on #11877.

closed 2022-10-03 by GregValiant