Ultimaker/Cura #13442: Slicing stalls at start

bneuen, 2022-10-02:
### Application Version

5.2.0-beta

### Platform

Linux 5.4.0-126-generic #142-Ubuntu SMP Fri Aug 26 12:12:57 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

### Printer

Ender 3

### Reproduction steps

Slicing stalls immediately, and must be cancelled, giving "Slicing failed with an unexpected error".

Slicing proceeds and completes properly on v5.1.1 with the same model and similar settings.

### Actual results

Slicing stalls immediately, and must be cancelled, giving "Slicing failed with an unexpected error".

### Expected results

Slicing should proceed and complete properly.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[slicing_stall_5.2.0-beta.zip](https://github.com/Ultimaker/Cura/files/9693691/slicing_stall_5.2.0-beta.zip)
[image]
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/9693693/cura.log)

GregValiant, 2022-10-03:
Thanks for the report.
I can't duplicate the problem on my Windows computer and the model sliced fine for me.  @MariMakes or someone else from the Cura team will take a look.

MariMakes, 2022-10-03:
Hey @bneuen,

Welcome to the Ultimaker Cura Github 🚀 .

I'm seeing the same as GregValiant, it slices on my Windows machine. 
But my colleague on Linux is seeing the same behavior as you are describing.

I'll bring it up to the team to discuss what we can do to resolve it. 
Thanks for reporting 💪

Piezoid, 2022-10-03:
Duplicate of #13422: AppImage lacks the libspdlog.so.1 shared object library.

MariMakes, 2022-10-03:
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13422
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13422

closed 2022-10-03 by MariMakes