Ultimaker/Cura #13443: crashing

TheFeralEngineer, 2022-10-03:
### Application Version

5.2.0-beta (and 5.1.0 stable)

### Platform

windows 11

### Printer

elegoo neptune 3

### Reproduction steps

opened program
opened part
sliced part
preview
crash

### Actual results

crashes if you slice a part more than once

### Expected results

no crashing

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Faulting application name: Ultimaker-Cura.exe, version: 0.0.0.0, time stamp: 0x6336e73e
Faulting module name: igxelpicd64.dll, version: 31.0.101.3358, time stamp: 0x62ec9ec0
Exception code: 0xc000041d
Fault offset: 0x0000000000080137
Faulting process id: 0x3814
Faulting application start time: 0x01d8d6c80c6570fd
Faulting application path: C:\Program Files\Ultimaker Cura 5.2.0-beta\Ultimaker-Cura.exe
Faulting module path: C:\WINDOWS\System32\DriverStore\FileRepository\iigd_dch.inf_amd64_f60bb83011df0c7e\igxelpicd64.dll
Report Id: 4234ac61-b76a-4618-b477-4fb00c57b03e
Faulting package full name: 
Faulting package-relative application ID:

MariMakes, 2022-10-19:
Hey @TheFeralEngineer ,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Do you still have the project file that causes the crash? 
It contains the printer and settings we need for troubleshooting. 

To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

TheFeralEngineer, 2022-10-19:
It was every file I used.

Open file
Import stl model
Slice
Do anything else... Crash

I just updated to 5.2.0 official release, so I'll let you know if it
happens again. It was happening in 5.1.0 and in 5.2.0 beta.

I'll try the models I know will crash and report back to you.

Thanks!

Phil T.
The Feral Engineer

Check out my LinuxCNC tutorials, machine builds and other antics at
www.youtube.com/c/theferalengineer

Help support my channel efforts and coffee addiction:
www.patreon.com/theferalengineer

Order one of the coolest label makers on the market at
http://labelworks.epson.com, use coupon code "theferalengineer" and receive
20% off of your order 🙂

TheFeralEngineer, 2022-10-25:
> Hey @TheFeralEngineer <https://github.com/TheFeralEngineer>,
>
> Can you let us know if you are still stuck?
> Thank you 😊

Hello,

It is happening less with 5.2.0, but it still crashes on occasion.

I'll see if i can get an event log from Windows when it happens again

Phil T.
The Feral Engineer

Check out my LinuxCNC tutorials, machine builds and other antics at
www.youtube.com/c/theferalengineer

Help support my channel efforts and coffee addiction:
www.patreon.com/theferalengineer

Order one of the coolest label makers on the market at
http://labelworks.epson.com, use coupon code "theferalengineer" and receive
20% off of your order 🙂

MariMakes, 2022-11-08:
Hey @TheFeralEngineer,

We don't have enough information to help you troubleshoot this issue right now. 
If you encounter this again, could you please create a new issue with the relevant information like project files and logs? 
Thank you 😊 

I'll be closing this issue. I hope you understand.

closed 2022-11-08 by MariMakes