Ultimaker/Cura #13445: Slicing Failed

Caratstix, 2022-10-03:
### Application Version

5.1.0

### Platform

Mac

### Printer

Creality CR10S Pro V2

### Reproduction steps

1. Loaded a Deadpool helmet file (from Do3D.com) into the slicer (230.7mb)
2. Setup helmet in preparation to slice and print
3. Pressed Slice Now button
4. Got error message saying Slicing Failed
5. Tried moving helmet, adjusting settings, no luck getting it to slice

### Actual results

unable to get Cura to slice file

### Expected results

Should slice file and allow me to save the GCODE to a flash drive for my printer

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-10-03:
Thanks for the report.
Without a project file it isn't possible to do an investigation.  You can try loading the file in Cura and using the "File | Save Project" command.  Zip the 3mf file and post it here.
That's the usual plan anyway.
The problem is your file is 230mb and I know my system won't be able to slice it and it may be above the size of file that you can upload here.
Have you checked that model for errors?  Using MS 3D Builder or Mesh Mixer or Cura's Mesh Tools may point to issues with the model.
I've been using 5.2beta since it's release but I haven't done enough slicing yet to know if the "won't slice" Windows bug has been repaired but so far it looks promising.

Caratstix, 2022-10-03:
[CCR10SPRO_MAIN DEADPOOL TEXTURED from Do3D.3mf.zip](https://github.com/Ultimaker/Cura/files/9701246/CCR10SPRO_MAIN.DEADPOOL.TEXTURED.from.Do3D.3mf.zip)

I did use the MeshTools to check for any errors and there were none present.

I can try to simplify the model however I am nervous to do this being that this file is specifically a very well detailed file that is supposed to be printed in red filament and not sanded/primed.

Thanks!b bv b bvv

Piezoid, 2022-10-03:
> CCR10SPRO_MAIN DEADPOOL TEXTURED from Do3D.3mf.zip

The .3mf file seems corrupted.

MariMakes, 2022-10-04:
Hey @Caratstix,

Thanks for your report. 👍 
I'm running into the same trouble as Piezoid and GregValiant.... I cannot open your file to help you troubleshoot. 

Can you try again to share your project file?
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

Caratstix, 2022-10-04:
Hi @MariMakes I was trying to attach a new project file last night but I could not get them under the 25MB allowed size.  If I were able to save the files to a public GoogleDrive folder and provide that link here would that work so you could look into it?  

I know the file I'm having issues with is an extremely detailed file of the Deadpool mask, its intended to be printed with red filament and needs no post processing work (no sanding, priming or painting) as the file is designed to look like leather fabric just like the mask from the films therefore I know its a bigger, more detailed file than I've uploaded/printed to date.

Thanks!

MariMakes, 2022-10-04:
I've had some good experiences with Wetransfer.
Do you mind uploading your file here? 
https://wetransfer.com/

You can share the link that you get in this issue.

Caratstix, 2022-10-04:
Hello @MariMakes @Piezoid @GregValiant I was able to upload the compressed file to a Google Drive link that I have here:

https://drive.google.com/file/d/1hL5itC4TUS5AbJGDwojbKIMK5jZj2_X7/view?usp=sharing

Hopefully you are able to access this file now!  Thank you all for your help thus far!!

Piezoid, 2022-10-04:
Thanks!
I can confirm that CuraEngine from the main branch crash on this model. Turning off vertices merging didn't help. 

The stl looks sane:
```
================= Results produced by ADMesh version 0.98.4 ================
Input file         : /tmp/DeadpoolFace.stl
File type          : Binary STL file
Header             : Uranium STLWriter Tue 04 Oct 2022 18:41:59
============== Size ==============
Min X = -127.189270, Max X =  124.929596
Min Y = -126.243576, Max Y =  124.603073
Min Z =  0.000000, Max Z =  175.668793
========= Facet Status ========== Original ============ Final ====
Number of facets                 : 4614292               4614292
Facets with 1 disconnected edge  :     0                   0
Facets with 2 disconnected edges :     0                   0
Facets with 3 disconnected edges :     0                   0
Total disconnected facets        :     0                   0
=== Processing Statistics ===     ===== Other Statistics =====
Number of parts       :     1        Volume   :  287209.750000
Degenerate facets     :     0
Edges fixed           :     0
Facets removed        :     0
Facets added          :     0
Facets reversed       :     0
Backwards edges       :     0
Normals fixed         : 4614288
```
Fixing the normals, and rotating the model along the Z axis didn't help either.

I was able to slice the model with increased integer precision (Ultimaker/CuraEngine#1710):
[image]

Caratstix, 2022-10-05:
Sorry this is all a bit new to me, so how would I go about getting this file to slice on cura so it can be printed? @Piezoid

MariMakes, 2022-10-05:
Hey @Caratstix,

Thanks for sharing your file. 
I checked for you and the file can be sliced in Cura 4.13.1.
[You can use an older version of Cura to get unstuck, ](https://github.com/Ultimaker/Cura/releases) but you'll miss out on some of the cool variable linewidth features. 
[image]

I'll bring your project file up to the team so they can learn what kind of error is preventing the slice from finishing.

Thanks for the report and good luck with printing! 👍 
Will you let us know how your print turns out? I'm super curious. 😍

Piezoid, 2022-10-05:
> Sorry this is all a bit new to me, so how would I go about getting this file to slice on cura so it can be printed? 

@Caratstix 
I used a custom patched version for CuraEngine to solve this recurrent issue with detailed models. Sadly I cannot distribute it as I don't have the infrastructure setup required for building it for Windows. Even less so for macOS, which is a gatekeeped platform hostile to independent developers.

Nevertheless, I have some reservations about that cubic texture coming out perfectly with traditional FDM printing. I think that if you simplify the mesh enough to remove the texture, Cura will be able to slice it. However I don't know how to do that without introducing artifacts.
As MariMakes stated, your best bet is to use 4.x versions of Cura, until this problem is resolved in 5.x.

Caratstix, 2022-10-24:
Hi @MariMakes 

I was able to download an older version of Cura and slice the file no problem, I ended up already printing 2 masks and the quality came out incredible.  This STL file was meant to mimic the look of fabric/leather and it really does look incredible.  Ive attached a couple pics of the finished product!
[image]
[image]
[image]

MariMakes, 2022-10-24:
[image]