Ultimaker/Cura #13446: Installation directory and application name should not include version numbers

fexofenadine, 2022-10-03:
**Application version**
All

**Platform**
Windows 11

**Printer**
N/A

**Reproduction steps**
1. Install version 5.1.1
2. Install version 5.2 Beta
3. Look at installed items in Start menu, Add/Remove Programs list, directories in c:\Program Files

**Actual results**
Upgrading this application on Windows is a nightmare because each installer installs a new package to a new unique directory, and has a new unique package name. It does not clean up previous versions and leaves orphaned files and packages everywhere

**Expected results**
Installer should not include version numbers to allow in-place upgrades without creating duplicate installs and orphaned items

**Project file**
N/A

**Log file**
N/A

**Additional information**
Could affect installers on other platforms too, I dont have experience with those

MariMakes, 2022-10-03:
Hey @fexofenadine,

Welcome to the Ultimaker Cura Github 🚀 

The behavior you are describing is actually intended behavior. 
Many users prefer running different versions of Cura next to each other. 
Especially with the significant slicing engine change, we introduced in Cura 5.0 and the fact our Beta versions potentially have breaking bugs in them. 

But you are pointing out correctly that we fail to enable you to remove your old version of Cura when installing a new version. We used to have a feature like that in our installing wizard but this has been removed without providing a proper alternative. 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9707

Thanks for the report! 👍

dsmaher, 2022-10-22:
IMHO, the preference to use side-by-side should be an option, and not the default. Or, at the *very least* patch-level and minor version upgrades should overwrite previous editions, but it is reasonable to default to side-by-side for major upgrades.  I can understand when major feature updates occur that you might want different versions, but it is _extremely_ annoying when 5.1.0 and 5.1.1 are different installations even though the updates are minimal, and for many users _not even relevant_.

I'd also like to point out that the uninstall of older versions also don't clean up after themselves, and "open with" in explorer generally is worthless since it might point to a version that you uninstalled *years* ago.  (since I'm a seasoned developer, I've manually cleaned this up on my machines _several_ times, but it is not for the feint of heart). 

Also, with the introduction of `winget`, the `winget upgrade` feature should do a complete overwrite of *any* existing version since that is the expected behavior.  Right now, `winget upgrade` will continually try install the latest version if any previous version also exists on the machine.

Basically CURA does not follow best practice for software upgrades, and this is probably causing _more_ severe problems than the ones it is intending to solve.