Ultimaker/Cura #13447: I cannot slice these objects

bjacek, 2022-10-03:
### Application Version

5.1.0, 5.2.0 beta

### Platform

Windows 10 Home

### Printer

Ultimaker 3

### Reproduction steps

1. I loaded STL file for one object into Cura.
2. I pressed Ctrl+M to create one more object.
3. I placed one of them using coordinates (-25, 0) and second one (25, 0)
4. I tried to slice'em.
5. Cura cannot finish slicing.
6. I moved one of the object to coordinates (-25, 5) and second one to (25, 5).
7. I tried to slice'em.
8. Cura worked perfect.
[UM3_X20 - shield.zip](https://github.com/Ultimaker/Cura/files/9697240/UM3_X20.-.shield.zip)

### Actual results

Slicing to the end of the world ;)
In Cura ver. 5.1.0 I can cancel slicing.
In Cura ver. 5.2.0 I cannot stop slicing (software stops to responding).

### Expected results

I should be able to send data to the printer.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[UM3_X20 - shield.zip](https://github.com/Ultimaker/Cura/files/9697258/UM3_X20.-.shield.zip)

GregValiant, 2022-10-03:
Thanks for the report and the project file.
I can duplicate this on my Windows computer and Cura 5.2beta.
I let the slicing run for a while but it did not complete and canceling caused Cura to crash.
Someone from the Cura team will look at this.
@MariMakes this file was not a problem with my Ender 3 Pro active but the OP's UM3 hates it.

MariMakes, 2022-10-05:
Hey @bjacek,

Quick update! The team found the cause of all the error messages and made a fix here: https://github.com/Ultimaker/CuraEngine/commit/e9396ff3a4fea0106f41ff77335d46cea65f1c7e

We expect this fix to prevent users with Cura version 5.2 stable and up to encounter this problem.  💪 

I'll close this issue since we expect it to be resolved, but if you are still encountering this, feel free to open it again. 
Thanks again for reporting 👍 🎉

closed 2022-10-05 by MariMakes