Ultimaker/Cura #13449: Inconsistent selection behavior after scrolling

ungutknut, 2022-10-03:
### Application Version

5.2.0-beta

### Platform

Windows 10

### Printer

doesn't matter

### Reproduction steps

1. Click on the textbox for any custom setting (e.g. Layer Height). A cursor appears after the text or number.
2. Scroll using the mouse wheel.
3. Click that custom setting again (or any other that features a text box). Once more a cursor appears after the textbox content.
4. Now use the scroll bar instead of the mouse wheel.
5. Click that custom setting again. This time the content gets highlighted.

### Actual results

Inconsistent selection behavior for custom settings when using scroll bar.

### Expected results

Consistent selection behaviour with "content highlighted" being default (less clicks than "cursor after content").

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Happens on Win10 and Linux versions.

MariMakes, 2022-10-03:
Hey @ungutknut,

Thanks for your report 👍 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/12227
You are the reporter of that issue! 🤣 
We still have a ticket on our backlog with the intent to improve this behavior.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12227

Unless you feel that this is a different issue, than we can open it again. 😉

closed 2022-10-03 by MariMakes

ungutknut, 2022-10-03:
> You are the reporter of that issue!

Hahaha... OMG, sorry. I was believing I only reported it on some UM forum feedback thread.