Ultimaker/Cura #13451: Printhead collides with printed object at end of job

youcangetme, 2022-10-03:
### Application Version

5.2.0-beta

### Platform

Windows 11

### Printer

Ultimaker S5

### Reproduction steps

1. Slice model using two colors of Tough PLA and an ooze tower
2. Print

### Actual results

At the end of the print job, the printhead appears to have collided with the printed object. I did not witness it but this morning I found the print platform all the way down and the glass plate moved to the right side of the build platform by about 3 inches and crooked. The metal hold down tab on the left was moved to the open position. The tab on the right had stop the glass plate from moving any further.

The printed object appears to be finished so this happened after the printing was complete. Probably during the filament retract.

When I opened the fan door on the print head it is clearly damaged by the collision. I was able to straighten the hinge somewhat. I could not see any damage to the belts. I will need to get out the calibration sticks to check it the head is no longer in alignment.

### Expected results

Complete the process of retracting the filament and return to the parked position without striking the printed object.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have the printed file in Digital Factory, I saved it out of the history. I am scared to run it again considering the damage the printer sustained.

Not sure how I can share it here.

I had already closed and shutdown the PC after starting the print last night.

MariMakes, 2022-10-03:
Hey @youcangetme,

Thanks for the report 👍 

I would love to learn more about your issue. I would need the file you printed with. 
Even if it's just an ufp, but preferably your project file. 
If it's the last thing you sliced and you open Cura, it should have remembered the last settings you used so you can reslice it and share the project file with us. 

To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

youcangetme, 2022-10-03:
[UMS5_ISO30 Tool Hook.zip](https://github.com/Ultimaker/Cura/files/9697893/UMS5_ISO30.Tool.Hook.zip)
Here is the project file. I tried to recreate the position but everything else is left untouched.