Ultimaker/Cura #13452: Initial start USB Cable vs Downloaded file

777BDVGP, 2022-10-03:
### Application Version

5.2.0-beta

### Platform

Windows 10

### Printer

Creality Ender 5 Plus.

### Reproduction steps

1 First, after slicing, print the object directly via a USB cable. (direct via USB cable)
2 Then with the exactly same print, down loaded the file to a Flash Disk for printing from the printer.  (Via download file)

### Actual results

1. With the first action with printing - The initial started at the corner, did the jet cleanup on the Y Axis on the side and reverse back to the corner.  100% as expected.
2. With the second action with printing selecting the file on the printer - The initial start did like the first (1), down to the Y Axis , but then when it was supposed to move right back to the corner, it actually first move to the middle of the print bed , and then back to the corner. ( see video attached. )

This problem is repeatable on any print.

https://user-images.githubusercontent.com/12105137/193577367-48573a45-ae19-42d5-881f-71395f7a7f4c.mp4

### Expected results

I expect that the print on the printer from the flash drive (2) will execute the same as directly from the PC via USB cable (1)
Zip file contain the Project file and the G-code file that was downloaded on the flash drive.
[Files.zip](https://github.com/Ultimaker/Cura/files/9697774/Files.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

None

777BDVGP, 2022-11-01:
> Thanks for the report.
> This behavior may be related to your firmware. If you have "Auto Bed Leveling" enabled in the printer AND the M420 line in the startup gcode then there can be a conflict and the purge lines don't go down correctly. The problem occurred with CR10's and Ender5's which (at one time) shared the same mainboard.
> You can try disabling Auto Bed Leveling in the printer by finding the menu item on the LCD, or in Cura, you can go to "Manage Printers / Machine Settings" and in the StartUp Gcode box put a semi-colon in front of the M420.
>
> As to why it is happening when printing via the USB and not when printing from the SD card I have no idea.
>
> I'll leave the bug tag on this for now, but it is likely a conflict between the M420 line in your StartUp Gcode and your firmware. Most Creality printers have moved away from M420 and use G29 instead.
>
> Let us know if you find a way around this. The line "M420 S1 Z2 ;Enable ABL using saved Mesh and Fade Height" is in your Ender 5 definition file which was written by "Trouch.com" and submitted to UltiMaker for inclusion in Cura.

Hi GregValiant,

 

I will look at your suggestions!

 

In the meantime, I have Fix/Bypass the problem by removing from the “Start G-code” the following line: (however it does not make sense)

G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little

 

It is as if this command is interpreted by the printer to move to the centre. Again, this is only needed for printing from the SD card. 

 

Well, I live with it, without this “ Move to side a little “ command.

 

Regards

Bennie Vermeulen

0824926676

GregValiant, 2022-11-01:
Well that seems to confirm that it's a conflict with the firmware.  I just reviewed the standard Ender 5 Plus definition file and I really don't get how that line could be a problem.  Creality uses almost exactly the same purge routine for all it's printers and they all include that jog.  It's there so the second purge line doesn't lay down where the first one already is.

I'll go ahead and close this then as you seem to have found a fix.  Nice job troubleshooting.

closed 2022-11-01 by GregValiant