Ultimaker/Cura #13453: 5.2 Beta Appimage Slicing not working

ckvsoft, 2022-10-03:
### Application Version

5.2 Beta Appimage

### Platform

Linux / Ubuntu 22,04

### Printer

Any

### Reproduction steps

load an stl file and go to slicing

### Actual results

none

### Expected results

sliced ​​object

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Unfortunately, I can't slice an object at all. It stands forever slicing but nothing changes. you can only cancel

[cura.log](https://github.com/Ultimaker/Cura/files/9698113/cura.log)

MariMakes, 2022-10-03:
Hey @ckvsoftm

Thanks for the report! 👍 
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13422
We have some developers working on the fix as we speak! 😤 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13422

closed 2022-10-03 by MariMakes

nallath, 2022-10-06:
I've uploaded a new version of the beta. Enjoy!

ckvsoft, 2022-10-09:
Working now. Thx