Ultimaker/Cura #13455: Save as .3mf in CURA5.2.0-beta and edit in 3D Builder, resizing of 10000%

Speedy-McCat, 2022-10-03:
### Application Version

5.2.0-beta

### Platform

Windows 10

### Printer

Anycubic Vyper

### Reproduction steps

1. An object is designed in the 3D builder and saved as an .stl file
2. Opened in Cura
3. saved as a project file (.3mf) in Cura
4. opened the .3mf file in the 3-Builder, called the repair and saved it again as a .3mf file
5. then opened this file in Cura and now I have a resizing of 10000%
6. When I open the .3mf file in 3D Builder, I see everything "normal"
[AV_Lichthalter für Roller 04.zip](https://github.com/Ultimaker/Cura/files/9699205/AV_Lichthalter.fur.Roller.04.zip)

### Actual results

opened this file in Cura and now I have 10000% resize

### Expected results

the object should be displayed in the cura on the push plate in normal size
[cura.log.zip](https://github.com/Ultimaker/Cura/files/9699253/cura.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log.zip](https://github.com/Ultimaker/Cura/files/9699274/cura.log.zip)

e-SonOfAnder, 2023-02-09:
I just encountered this same bug in 5.2.1 with a 3MF file that I exported from OnShape. The imported item is at the correct size, but the scale claims to be 10,000.01%, as shown in this image:
[image]

Attached is the 3MF file, and the only similarity that I can see here is that I also had to rotate the object to get it flat to the build plane, because it was made in the XZ plane in OnShape instead of the XY plane. I can try rotating the object in OnShape and re-exporting it to see if it triggers the same bug in Cura.
[DC x DA.zip](https://github.com/Ultimaker/Cura/files/10694182/DC.x.DA.zip)

e-SonOfAnder, 2023-02-09:
Just tested by rotating the object in OnShape and exporting again, and it made no difference; it still imports at the right size, but claiming to be at 10,000.01% scale.

GregValiant, 2023-02-09:
I don't know if that is the same issue.  The original issue I think is with MS 3D Builder as other 3mf files come into Cura OK.
Does the OnShape STL exporter have any settings?

GregValiant, 2023-02-09:
I've been experimenting a bit.
When I save a Cura project file it's a 3mf file.  When I open that in MS 3D Builder it's all good.  When I save that file in MS 3D Builder and then open it Cura it gets scaled up 10000%.  So from what I see the bug may be in the 3mf file that Cura creates OR it may be in the one that comes back from MS 3D Builder.  I think that's what the OP noticed as well as he was designing in MS 3D.
When I import check a project file model in MS3d and then save it and bring it back into Cura - although the XYZ are scaled up 10000% the "percentage" boxes show "393.7%" so something is sideways there.  I can set the model back to it's original size (if I happen to know it) but not to "100%" which should be the same thing.
I don't think it's a coincidence that 1/25.4 = .003937 and that 3.937% scaling also happens to be the percentage that is correct for the model returned from MS 3D Builder.

I open a lot of project files from here and from the UM Forum.  A large percentage of the problems that people have are actually model problems and I've found MS 3D Builder to be a pretty good tool.  I have noticed the strange behavior when bringing models back into Cura.  It's a "chicken and egg" kind of bug.  The Cura team has far better tools to look into things like this and it's their call anyway.