Ultimaker/Cura #13456: CUra 5.2.0-beta (`monotonic top/bottom order` broken)

discip, 2022-10-03:
### Application Version

5.2.0-beta

### Platform

WIndows 10

### Printer

does not matter

### Reproduction steps

1. Load the attached file
2. slice it with **`monotonic top/bottom order`** activated
3. watch the the print head traveling for each top/bottom line around the inner perimeter instead of filling in these lines consecutively on each side (see attached gif)

### Actual results

5.2.0-beta:

https://user-images.githubusercontent.com/53649486/193637838-506425e2-65fd-4e50-bc5e-317e95653c24.mp4

### Expected results

5.1.1:

https://user-images.githubusercontent.com/53649486/193637938-308904b6-c9f9-4b82-abae-c2a4f3fbae2f.mp4

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[project.zip](https://github.com/Ultimaker/Cura/files/9699492/project.zip)

Guilouz, 2022-10-03:
Yes I confirm this issue.

MariMakes, 2022-10-04:
Hey @discip and @Guilouz,

Thanks for the quick report 👍 

You are correct, it's broken. 
[image]

We have someone on the team working on it to resolve it for the stable version of 5.2.
For internal reference CURA-9710  🤞

Piezoid, 2022-10-05:
@rburema: This is caused by Ultimaker/CuraEngine@e37ec592

richfelker, 2022-10-05:
I hit this bug today, and have just reverted the above commit locally. But it'd be nice to have an analysis of what the right fix is.

rburema, 2022-10-06:
@Piezoid Thanks! We know, a colleague has picked it up during my long weekend. We'll figure out the original problem and can revert that one.

Vandresc, 2022-10-14:
Fixed in 5.2 stable.

closed 2022-10-14 by Vandresc