Ultimaker/Cura #13459: Documentation of Material Configurations

julianCast, 2022-10-03:
### Is your feature request related to a problem?

Currently, the only printer capable of showing Material Configurations UI is the Ultimaker S5, having a documentation would allow developer, as me, to implement it for any other printer.

[image]

### Describe the solution you'd like

I'd like the documentation of Material Configurations: What is the object returned when the S5 is added from the cloud enabling the material configurations? E.j.:
{printer : {id, model}
materials : [{id}, {id}]}

### Describe alternatives you've considered

None yet.

### Affected users and/or printers

Any user with a material multiload system. Either custom or branded.

fieldOfView, 2022-10-03:
These configurations require the printer somehow communicating with Cura what material and nozzle size is currently configured on the printer. The Ultimaker 3, S5, S3 and 2+ Connect all communicate with Cura through a bespoke protocol, either via the Ultimaker Cloud or locally (AFAIK, the latter is not available with the 2+ Connect; that printer only supports the cloud connection). The Cura-part of that bespoke communication can be found here: https://github.com/Ultimaker/Cura/tree/main/plugins/UM3NetworkPrinting

You are going to have to write (or extend) a plugin that communicates with the printer. That plugin will have a subclass of the PrinterOutputDevice, and that subclass needs to implement a uniqueConfigurations property which returns a list of PrinterConfigurationModel instances.

julianCast, 2022-10-04:
Thank you for your answer. 

I do have a working subclass of `PrinterOutputDevice` as a plugin. But I have some questions:
- Where/When `self._unique_configurations` is set ? [Here ?](https://github.com/ultimaker/cura/blob/main/cura/PrinterOutput/PrinterOutputDevice.py#L257) As I can't see the flow, I don't know if the cloud printers come with the configurations data or if it's retrieved later.
- What is the difference between  `printer.printerConfiguration` and `printer.availableConfigurations` ? [This.](https://github.com/BCN3D/Stratos-intern/commit/d1720db5ad92d9832cf4706d87caf96550c90fef)

fieldOfView, 2022-10-05:
> Where/When `self._unique_configurations` is set 

That is entirely up to your implementation. Your plugin should somehow receive information from the printer/collection of printers, then populate a list of `PrinterConfigurationModel` instances with that data and emit the `uniqueConfigurationsChanged` signal.

> What is the difference between `printer.printerConfiguration` and `printer.availableConfigurations`

`printerConfiguration` is the active configuration (material, variant and buildplate) of the current printer. The `availableConfigurations` are configuration options that a printer can switch to, but doesn't currently have active (configurations that can be activated with eg; automatic tool changes, material loaders, etc). Both of these descriptions can be found in the `PrinterOutputModel` source.

julianCast, 2022-10-06:
Hi! So, to summarize:
- **printerConfiguration**: The active configuration (material, variant and buildplate) of the current printer. The one users have selected.
- **availableConfigurations**: Configuration options that a printer can switch to, but doesn't currently have active. The ones I'll set when I retrieve the data.
- **uniqueConfigurations**: Automatically calculated configurations based on the availableConfigurations changes. Shown in the UI. 

As I understand, **availableConfigurations** would contain something like ext1: material1 & empty, ext2: material1 & material2 and **uniqueConfigurations** would be the calculation result of all the combinations (the ones shown in the UI to pick).