Ultimaker/Cura #13461: Installation Issue

XeroZar, 2022-10-03:
### Application Version

5.2.0 Beta

### Platform

Windows 10 Pro

### Printer

Ender 3 Pro

### Reproduction steps

Downloaded and installed Ultimaker - Cura 5.2.0 beta from link thru app. Installed but keeps asking to update to Beta 5.2.0 and the old version of cura wasn't removed...

### Actual results

Still keeping old version and doesn't recognize the new version installed

### Expected results

Older version should not be there anymore, all settings imported to new installed version
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

None

MariMakes, 2022-10-04:
Hey @XeroZar ,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for the report. 👍 
This sounds familiar 🤔 

We've had reports about this before, and it should have been resolved with this change:
https://github.com/Ultimaker/Uranium/pull/816

I'll bring it up to the team to see what might be going wrong. 
Fingers crossed 🤞

MariMakes, 2022-10-06:
Hey @XeroZar ,

Quick update on our side. 
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9732

Thanks for the report! 👍

XeroZar, 2022-10-06:
Hello ok thank you.

XeroZar, 2022-10-26:
Hello  again. So the new update guess fixed that issue. Now I have a new one... When I installed the new version 5.2.1, it went fine, but did not pull any of the settings. Unfortunately I didn't check before uninstalling previous version. Anyway of getting it back?