Ultimaker/Cura #13465: Speed drop on corners  - Overlap settings? - Beta 5.2

PhilBaz, 2022-10-04:
### Application Version

5.2 Beta

### Platform

win 10

### Printer

Voron 2.4

### Reproduction steps

Slice using overlapping modifier

[image]

### Actual results

image

### Expected results

constant/requested speed

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[speed drop on corners.zip](https://github.com/Ultimaker/Cura/files/9702528/speed.drop.on.corners.zip)

MariMakes, 2022-10-04:
Hey @PhilBaz,

Thanks for the report 👍 

I took a look, there is indeed something weird happening with the cutting mesh slowing down the printjob. 
Even if you don't change any of the settings which is odd. 
I'll have to bring it up with the team to analyze what's happening. 

It also seems to be happen in Cura 5.1, so it's not a Beta only bug. 
[image]

MariMakes, 2022-10-06:
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9731

[image]

Thanks for the report! 👍

PhilBaz, 2022-11-08:
Do you think this will get attention for Cura 5.3 ?  Its a big issue.

PhilBaz, 2023-01-05:
@MariMakes  Great. Do you think this will get attention for Cura 5.3?  It would be great to have overlap settings working. Very useful feature.

MariMakes, 2023-01-06:
Hey @PhilBaz,

I checked for you. 
Unfortunately, A fix for this will not make it into the 5.3 release. 
But there is some other cool stuff lined up!

PhilBaz, 2023-01-06:
@MariMakes 

Okay, thanks for letting me know... I'm looking forward to 5.3. 

But if I'm being honest I'm mostly using smart avionics master branch largely because Cura 5 just has too many bugs at this point. I would wish for a a focus on bug fixes rather than new features.  I seem to run into fundamental problems more often than not. So new features seem frivolous. 

That being said I appreciate all the work and effort, and look forward to using the 5 branch in the future.

PhilBaz, 2023-06-17:
Just wanted to bump this, as along with the stuttering issue this is probably the second biggest bug in all of Cura right now.

PhilBaz, 2023-06-17:
@MariMakes see above. Maybe for 5.4?

MariMakes, 2023-07-10:
Hey @PhilBaz,

I apologize that this issue does not have the priority that you want. 
We did work on [other mesh-related issues](https://github.com/Ultimaker/Cura/issues/15836), but unfortunately not this one. 

We see that only 1 in 500 slices are done with any type of meshes, that might be why it keeps dropping in priority. 
I'll raise it again with the team to ask if they can take another look.

PhilBaz, 2023-07-12:
@MariMakes 

Hi Mari,

No apology necessary of course. Sorry to be a PITA.

I see the stuttering mesh issue was solved and I participated in the thread and verified that it solved the problem on my prints to try and contribute what I can.

The reason I single this issue out in action to that one is because a core/primary functionality is broken/bugged. 

I admit I'm surprised to find out that utilization is not higher, and so I see the point.

Thanks for the reply. Ill keep my fingers crossed for the next release.