Ultimaker/Cura #13466: Error slicing file

klmullen, 2022-10-04:
### Application Version

5.1.0

### Platform

MacOS 11.17

### Printer

Ender 2 Pro

### Reproduction steps

Nothing special
 I loaded an STL file for "Waratah Bowl" from 3DPrintBunny. I selected the Low Quality (0.28 Layer Height) Slicing profile. I got an error trying to slice it. I tried with the Standard Quality profile (0.20 Layer Height) with the same result

### Actual results

[image]

### Expected results

[CE2_PLA_04_Waratah_Bowl_LOWQ.3mf.zip](https://github.com/Ultimaker/Cura/files/9702730/CE2_PLA_04_Waratah_Bowl_LOWQ.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE2_PLA_04_Waratah_Bowl_LOWQ.3mf.zip](https://github.com/Ultimaker/Cura/files/9702733/CE2_PLA_04_Waratah_Bowl_LOWQ.3mf.zip)

MariMakes, 2022-10-04:
Hey @klmullen,

Thanks for your report. That's a wonderful model! 😍 

[image]

I'm getting an error that something is wrong with your model. 
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools
If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/.

klmullen, 2022-10-05:
The model is by 3DPrintBunny. https://thangs.com/designer/3dprintbunny/3d-model/Waratah%2520Bowl-180294?part=3f1b703e-d6fe-4c84-84d3-714a5b872c73

 I’ve sliced it successfully with an earlier version of Cura. I think it was the last release of version 4. I’ll try the mesh fixes and see if that helps.

MariMakes, 2022-10-05:
Hey @klmullen,

I took a look and you are correct.

This model slices for me in Cura 4.13.1 but not in 5.2 Beta. 
[image]

So if you use Cura 4.13.1 you'll be able to slice and print your model.
I'll ask the team to check why it can't be sliced in Cura 5.x.

Thanks for the report 👍

Piezoid, 2022-10-05:
This is the recurrent "edge without a twin" error on detailed models. (`quad_end->twin = nullptr` when it segfaults [here](https://github.com/Ultimaker/CuraEngine/blob/main/src/SkeletalTrapezoidationGraph.cpp#L304)).

I was able to slice it with increased integer precision, which shows that the issue is likely related to very small polygons or near intersection in them.

ThomasRahm, 2022-10-07:
@Piezoid I also had segfaults [here](https://github.com/Ultimaker/CuraEngine/blob/main/src/SkeletalTrapezoidationGraph.cpp#L288) with `quad_mid->twin->prev = nullptr` 
I do not think that it only is dependent on the model, as my workaround is to move the model on the buildplate around until the error does not occur anymore.

Piezoid, 2022-10-07:
> @Piezoid I also had segfaults [here](https://github.com/Ultimaker/CuraEngine/blob/main/src/SkeletalTrapezoidationGraph.cpp#L288) with `quad_mid->twin->prev = nullptr` I do not think that it only is dependent on the model, as my workaround is to move the model on the buildplate around until the error does not occur anymore.
> 
> Edit: What i described above happened with another model, not the one linked here.

@ThomasRahm I agree, it is very sensible to the translation, rotation and layer height. That's is why fiddling with the model transformations often gets recommended on these tickets. PrusaSlicer's port of Arachne tries to solve this issue by programmatically rotating the model when something wrong is detected. ([references here](https://github.com/Ultimaker/CuraEngine/pull/1731#discussion_r970060091))
The issue is more likely to manifest itself on detailed models. IME, with finer details it becomes harder to find a working transformation. 
If I'm not mistaken, this error happens when `boost/polygon/voronoi` encounters near intersections and generates a diagram that violates invariants assumed by the skeletal trapezoidation (the presence of twin edges). Something was recently fixed in the discretization of parabola (https://github.com/Ultimaker/CuraEngine/pull/1737) but it doesn't help on this model.

Other solutions could be a combination of:
- checking these invariants more robustly (don't crash, worst case scenario: omit polygon region), 
- repairing the graph a posteriori (Ghostkeeper had a PR doing that, but it didn't get merged),
- filtering/fixing the geometries that cause the issues.

I also experimented with a patch enabling finer resolution for cura's coordinate system (Ultimaker/CuraEngine#1710) which resolves all the cases I was able to find here. (That's the reason why I'm following these tickets.) This is more like band-aid than a real solution, as I don't even know exactly why it works, and if it will work for all models. Near-intersections probably become less of an issue when snapping vertices to a finer grid.

closed 2023-01-08 by klmullen