Ultimaker/Cura #13467: Missing layers

RobertDarmond, 2022-10-04:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Sidewinder X1

### Reproduction steps

Slice an object. STL file from Fusion 360

### Actual results

Layers was missing.
On the preview and when printing.

I try many times. I quit and reload Cura.
This append at random layers, in the base of the object.

gCode show no layer 3 (can't find "LAYER:3" in file) and Z jump from 0.36 to 0.60 (layer height is 0.12) :

G0 F300 X156.752 Y149.285 Z0.24
...
;LAYER:1
...
G0 F300 X156.66 Y150.733 Z0.36
...
;LAYER:2
...
G0 F300 X156.892 Y148.803 Z0.6
...
;LAYER:4

[image]

When showing preview, sometimes all the layers are shown plus lines going to the origin. On the picture below, the slider is on layer 18. Layers 17 and 19 show correctly.
[image]

### Expected results

All layers printed.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Pique-Mais v4.zip](https://github.com/Ultimaker/Cura/files/9702791/Pique-Mais.v4.zip)

MariMakes, 2022-10-04:
Hey @RobertDarmond,

Welcome to the Ultimaker Cura Github 🚀 

I took a look at the file that you shared, and I can reproduce your empty layers.
But it does not seem to be an error in Cura. 

It seems that your mesh is broken.
The Mesh Tools Plug-in is indicating that your model is not watertight. 
You can get the plug-in here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

Do you still have the model in Fusion 360? 
Can you inspect the model in the part that shows the empty layers?
Can you check your export settings?

RobertDarmond, 2022-10-05:
Hello!
Thank you for your prompt response.

I have drop (and rebuild) the original model in Fusion. I only have the .stl file I sent to you.

If there is a problem in the model, I think it is not in the part that shows the empty layers in the base (witch is very simple) but in the other side of the object, near the pointed end. See layers 474 and 493.

You said "it does not seem to be an error in Cura".
But Cura generated a .gcode file with no warning and the print failed.
Fortunately, this print was not a very big one.
Cura should warn user. That is why I opened this issue.

What do you mean when you ask to "check the export settings" ?

GregValiant, 2022-10-07:
This is the view in Mesh Mixer.  The magenta colored areas show errors.  In this case there are only 2 errors but they include the whole model.
[image]

Here is the model sliced.  The model on the right has been repaired in 3D Builder.  You can see that the layer is correct where the unrepaired model shows a floor there because a surface normal in inside out.
[image]

I don't see a bug here as the model is in error.  Once repaired the model slices as it should.  I'll leave the bug label on for now and you can respond here.

madhephaestus, 2023-10-03:
For reference, i just tested this model in Cura 5.4.0, and it slices correctly.

GregValiant, 2023-10-03:
@RobertDarmond if you load the Mesh Fixes Plugin from the MarketPlace a warning should come up.
Can we close this??

closed 2023-10-18 by github-actions[bot]