Ultimaker/Cura #13475: Option to Change Priming Feature Print Order (prime tower or ooze shield)

Erin-Stephan, 2022-10-04:
### Is your feature request related to a problem?

Currently for dual head prints with priming features (prime tower or ooze shield), the printing sequence for each toolhead goes Prime Tower -> Part Layer A -> Prime Tower -> Part Layer B -> Tool Change. This adds extra travel moves away from the part that make more opportunities for stringing and makes the "A" layer cool more before the "B" layer starts.

### Describe the solution you'd like

I would like the option to select the order of Prime Tower Layer A -> Part Layer A -> Part Layer B -> Prime Tower Layer B -> Tool Change. This way, Part Layers A and B are sequential, thus reducing travel moves (and therefore stringing), and reducing the layer time between those two layers which helps with strength. 

An additional benefit is that when using the cool down speed settings, normally the cool down is at the end of part layer B which could cause jamming or reduced strength in that location from cooled material. With prime tower layer B at the end of the sequence, the final print temp and cooled material is more-so in the prime feature, which allows your part to have better strength and less opportunity for defects. 

### Describe alternatives you've considered

Alternative is the current solution, which technically works but has room for improvement as described above.

### Affected users and/or printers

This is most applicable to all dual head Cura users, specifically those that use the priming features such as the prime tower and ooze shield. The old functionality does not need to be lost, but the option could give users more selection in toolpathing.

### Additional information & file uploads

Images for current and proposed feature order.
[PrimingFeatureOrder.pdf](https://github.com/Ultimaker/Cura/files/9708777/PrimingFeatureOrder.pdf)

Erin-Stephan, 2022-10-17:
Can I get an update on this? This would be a really great feature!

ZacharyHowell-Essentium, 2022-10-17:
@nallath, @MariMakes

MariMakes, 2022-11-01:
Hey @Erin-Stephan,

Sorry, it took us a while to get back to you 😞 
Our Print Process Engineers took a look at your request. 
They agree that your suggestion is an improvement over the current behavior. 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9825

Thanks for the report! 👍

MariMakes, 2023-01-17:
Hey @Erin-Stephan,

Quick update on this issue.

Erin-Stephan, 2023-02-17:
Hi @MariMakes, my team wanted to test these settings early so we did a cura build from source (Cura 5.4.0-alpha) and there is a bug that I think this new order has introduced. On the sections that are meant to be all support, the extruder is alternating between model (T0) and support (T1). As the part is printing, the order goes T0 prime tower, T0 layer1, T0 layer 2, T0 prime tower, **T0 layer 2 (printing on the support path), then tool change T1 prime tower**, T1 Layer 3, T1 prime tower. That T0 layer 2 should be the T1 support extruder and that should come after the T1 prime tower. I hope this makes sense. I have attached the gcode (As a txt) and an image of the part. I realize this is an alpha, build from source so it may be taken care of or maybe an error as a result of this source build but I wanted to bring it to your attention. Thanks
[image]
[EHSE2_ComplexSupportTest.TXT](https://github.com/Ultimaker/Cura/files/10771122/EHSE2_ComplexSupportTest.TXT)

rijkvanmanen, 2023-02-22:
Thanks for reporting @Erin-Stephan. I made a quick fix for the issue in https://github.com/Ultimaker/curaEngine/tree/CURA-9825-improved-printing-order-of-prime-tower. I did not do any thorough testing yet, so let me know if you run into any issues.