Ultimaker/Cura #13477: can't slice

beemerben, 2022-10-05:
### Application Version

5.2 beta

### Platform

mac 

### Printer

ender 3

### Reproduction steps

every time i try to slice anything it gives me an error saying "slicing had unexpected error" 

### Actual results

every time i try to slice anything it gives me an error saying "slicing had unexpected error" 

### Expected results

it slicing. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[CE3PRO_chest left top 1.3mf.zip](https://github.com/Ultimaker/Cura/files/9712017/CE3PRO_chest.left.top.1.3mf.zip)
[image]

GregValiant, 2022-10-05:
There is no reason to yell.
The model has errors as described below.  
--> 0 Naked edges
--> 0 Planar holes
--> 0 Non-planar holes
**--> 5 Non-manifold edges
--> 8 Inverted faces
--> 4 Degenerate faces
--> 6 Duplicate faces**
--> 0 Disjoint shells
-> Repairing: 100.00%
----- Repair completed in 81024ms ------
-> Vertex count changed from 221727 to 183480 **(-38247)**
-> Triangle count changed from 443434 to 367148 **(-76286)**

When a model isn't watertight and/or has inverted faces then Cura cannot tell what is the inside and what is the outside.
This is what the repaired model looks like sliced.
[image]

I'll leave the bug label on but the model has some severe errors that are at least part of the problem.

beemerben, 2022-10-05:
Is there a method that you used to repair? I have tried several times but cura doesn’t have that feature Sincerely,Benjamin WellsBusiness Development(804) 363-6203

GregValiant, 2022-10-05:
The report was generated and the repair was effected by [OnLineSTLRepair](https://formware.co/OnlineStlRepair).
Cura has a utility "Mesh Tools" that you can load in the MarketPlace.  It's pretty good but not as full featured as a dedicated repair program.  The on-line site has a file size limit (20mb??) so it can't be used for everything.  Mesh Mixer can also repair models.

jellespijker, 2024-01-14:
Thank you for reporting this issue, I can confirm that the project file still crashes with our latest internal build 5.7.0-alpha.0+3abb55 . For internal reference the crash is logged crash under Sentry issue: [CURAENGINE-1X](https://ultimaker-o7.sentry.io/issues/4863502807/)